rtlopez/esp-fc #4: error

betmensk, 2019-02-19:
I'm sorry, but this project seems like one big red error :-(

c:\users\majo\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\random.tcc:1644:28: error:
 call of overloaded 'log(double)' is ambiguous

lib\MPU6050\src/helper_3dmath.h:74:54: error: 'sqrt' was not declared in this scope
lib\MPU6050\src/helper_3dmath.h:154:46: error: 'acos' was not declared in this scope
lib\MPU6050\src/helper_3dmath.h:165:48: error: 'sin' was not declared in this scope
lib\MPU6050\src/helper_3dmath.h:174:35: error: there are no arguments to 'acos' that depend on a template parameter, s
o a declaration of 'acos' must be available [-fpermissive]
lib\Espfc\src/Filter.h:80:31: error: 'M_PI' was not declared in this scope
etc...
Environment d1_mini             [ERROR]
 :-(

mahdiyousofun, 2020-03-06:
Also, I have this problem and I don't know what should I do.

yazdanv, 2020-03-30:
i also have this problem too, i think its for the new version of esp or pio

rtlopez, 2020-06-17:
fixed in v0.1.0-alpha5

closed 2020-06-17 by rtlopez