rtlopez/esp-fc #6: "Math.h" and "math.h" confused by Visual studio code

thijses, 2020-06-15:
i was trying to compile the V0.1.0 alpha and it kept failing to include "math.h", so i renamed your "Math.h" (and its namespace) and now it works. I am compiling from platformIO from MS Visual Studio Code (because i dont like AtomIO), so it seems like VScode just can't distinguish between "Math" and "math".

rtlopez, 2020-06-15:
Hi @thijes, I don't think it is good change. math.h and Math.h are two diferrent things you shouldn't mix them. The first one is intended to be imported from compilers library. 

can you run `pio run -e d1_mini -t clean` and then `pio run -e d1_mini -v` and paste the result here?

quite long time ago I've migrated to vscode, but didn't updated docs so far. Never the less I don't get any kind of error you mention. 

I think it might be because I'm developing on Linux, are you using Windows? It might be possible the source of problem, whilst Windows is not case-sensitive in file names.

thijses, 2020-06-15:
I am very aware of the different roles of the math.h and Math.h files.
I am indeed using Windows, yes, that's probably the cause.

(after running `pio run -e d1_mini -t clean`)
the VScode terminal after running `pio run -e d1_mini -v`:
(the terminal has too many warnings & errors, and starts to remove the top lines after a few seconds of compiling, but it's basically all errors about the same thing)

> In file included from lib\Espfc\src/Espfc.h:11:0,
                 from src\main.cpp:5:
lib\Espfc\src/Mixer.h: In member function 'void Espfc::Mixer::writeOutput(float*, size_t)':
lib\Espfc\src/Mixer.h:215:141: error: 'lrintf' was not declared in this scope
             const int16_t tmp = lrintf(Math::map3(out[i], -1.f, 0.f, 1.f, och.reverse ? 2000 : 1000, och.neutral, och.reverse ? 1000 : 2000));
                                                                                                                                             ^
lib\Espfc\src/Mixer.h:221:122: error: 'lrintf' was not declared in this scope
             _model.state.outputUs[i] = lrintf(Math::map(v, -1.f, 1.f, _model.state.minThrottle, _model.state.maxThrottle));
                                                                                                                          ^
In file included from lib\Espfc\src/Device/GyroDevice.h:4:0,
                 from lib\Espfc\src/ModelConfig.h:7,
                 from lib\Espfc\src/Model.h:12,
                 from lib\Espfc\src/Espfc.h:4,
                 from src\main.cpp:5:
lib\AHRS\src/helper_3dmath.h: In instantiation of 'VectorBase<T> VectorBase<T>::eulerFromQuaternion(const Quaternion&) [with T = float]':
lib\Espfc\src/Fusion.h:104:47:   required from here
lib\AHRS\src/helper_3dmath.h:362:69: error: 'atan2f' was not declared in this scope
       x = atan2f(q.w * q.x + q.y * q.z, 0.5f - q.x * q.x - q.y * q.y);
                                                                     ^
lib\AHRS\src/helper_3dmath.h:363:48: error: 'asinf' was not declared in this scope
       y = asinf(-2.0f * (q.x * q.z - q.w * q.y));
                                                ^
lib\AHRS\src/helper_3dmath.h:364:69: error: 'atan2f' was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
       z = atan2f(q.x * q.y + q.w * q.z, 0.5f - q.y * q.y - q.z * q.z);
                                                                     ^
lib\AHRS\src/helper_3dmath.h:362:69: note: 'atan2f' declared here, later in the translation unit
       x = atan2f(q.w * q.x + q.y * q.z, 0.5f - q.x * q.x - q.y * q.y);
                                                                     ^
lib\AHRS\src/helper_3dmath.h: In instantiation of 'VectorBase<T> VectorBase<T>::accelToEuler() const [with T = float]':
lib\Espfc\src/Fusion.h:110:59:   required from here
lib\AHRS\src/helper_3dmath.h:308:33: error: 'atan2f' was not declared in this scope
       rpy.x =  atan2f(na.y, na.z);
                                 ^
lib\AHRS\src/helper_3dmath.h:309:59: error: 'sqrt' was not declared in this scope
       rpy.y = -atan2f(na.x, sqrt(na.y * na.y + na.z * na.z));
                                                           ^
lib\AHRS\src/helper_3dmath.h:309:59: note: suggested alternative:
In file included from c:\users\tttth\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\random:38:0,
                 from c:\users\tttth\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:65,
                 from c:\users\tttth\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62,
                 from C:\Users\tttth\.platformio\packages\framework-arduinoespressif8266@3.20603.200130\cores\esp8266/Arduino.h:237,
                 from src\main.cpp:1:
c:\users\tttth\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\cmath:494:5: note:   'std::sqrt'
     sqrt(_Tp __x)
     ^
In file included from lib\Espfc\src/Device/GyroDevice.h:4:0,
                 from lib\Espfc\src/ModelConfig.h:7,
                 from lib\Espfc\src/Model.h:12,
                 from lib\Espfc\src/Espfc.h:4,
                 from src\main.cpp:5:
lib\AHRS\src/helper_3dmath.h:309:60: error: 'atan2f' was not declared in this scope, and no declarations were found by argument-dependent lookup at the point of instantiation [-fpermissive]
       rpy.y = -atan2f(na.x, sqrt(na.y * na.y + na.z * na.z));
                                                            ^
lib\AHRS\src/helper_3dmath.h:308:33: note: 'atan2f' declared here, later in the translation unit
       rpy.x =  atan2f(na.y, na.z);
                                 ^
lib\AHRS\src/helper_3dmath.h: In instantiation of 'Quaternion VectorBase<T>::eulerToQuaternion() const [with T = float]':
lib\Espfc\src/Fusion.h:129:66:   required from here
lib\AHRS\src/helper_3dmath.h:342:34: error: 'cosf' was not declared in this scope
       float cosX2 = cosf(x / 2.0f);
                                  ^
lib\AHRS\src/helper_3dmath.h:343:34: error: 'sinf' was not declared in this scope
       float sinX2 = sinf(x / 2.0f);
[... 47 lines cut ...]
                 from lib\Espfc\src/Model.h:12,
                 from lib\Espfc\src/Espfc.h:4,
                 from src\main.cpp:5:
lib\Espfc\src/Filter.h: In member function 'float Espfc::Filter::getNotchQ(int, int)':
[... cut ...]
lib\AHRS\src/helper_3dmath.h:233:5: warning: control reaches end of non-void function [-Wreturn-type]
     }
     ^
In file included from lib\Espfc\src/ModelConfig.h:9:0,
                 from lib\Espfc\src/Model.h:12,
                 from lib\Espfc\src/Espfc.h:4,
                 from src\main.cpp:5:
lib\Espfc\src/Device/BaroDevice.h: In member function 'float Espfc::Device::BaroDevice::getAltitude(float, float)':
lib\Espfc\src/Device/BaroDevice.h:45:5: warning: control reaches end of non-void function [-Wreturn-type]
     }
     ^
*** [.pio\build\d1_mini\src\main.cpp.o] Error 1

rtlopez, 2020-06-15:
definetly it is result of not included math.h http://www.cplusplus.com/reference/cmath/, try with my fix in tag v0.1.0-alpha2.

rtlopez, 2020-06-17:
fixed in v0.1.0-alpha2

closed 2020-06-17 by rtlopez