rtlopez/esp-fc #32: wemos d1 mini can't connect betaflight configurator

AutoPlantBali, 2023-06-23:
[image]

rtlopez, 2023-06-24:
Hi @AutoPlantBali 

thanks for your report, can you provide me a firmware version you're using?

Are you able to connect to cli (115200 8N1) with alternative terminal, like PlatformIO, Arduino IDE, Putty, or Miniterm and type `status` command twice or more times?

AutoPlantBali, 2023-06-24:
thanks for replying..

i'am use visual code + PlatformIO, this result after click upload and monitor:

Processing d1_mini (board: d1_mini; platform: espressif8266; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.0.1) > WeMos D1 R2 and mini
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Espfc
|-- betaflight       
|-- EEPROM @ 1.0     
|-- EscDriver        
|-- ESP8266WiFi @ 1.0
|-- EspGpio
|-- EspSoftSerial    
|-- EspWire
|-- Kalman
|-- AHRS
|-- printf
|-- SPI @ 1.0        
|-- Wire @ 1.0       
Building in release mode
Compiling .pio\build\d1_mini\src\main.cpp.o
Generating LD script .pio\build\d1_mini\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\d1_mini\lib69c\printf\printf.c.o
Archiving .pio\build\d1_mini\lib69c\libprintf.a
Compiling .pio\build\d1_mini\libcd2\betaflight\blackbox\blackbox.c.o
Compiling .pio\build\d1_mini\libcd2\betaflight\blackbox\blackbox_encoding.c.o
Compiling .pio\build\d1_mini\libcd2\betaflight\blackbox\blackbox_io.c.o
Compiling .pio\build\d1_mini\libcd2\betaflight\platform.c.o
Archiving .pio\build\d1_mini\libcd2\libbetaflight.a
Compiling .pio\build\d1_mini\lib7f9\EscDriver\EscDriverEsp8266.cpp.o
Compiling .pio\build\d1_mini\lib7f9\EscDriver\EscDriverRP2040.cpp.o
Archiving .pio\build\d1_mini\lib7f9\libEscDriver.a
Compiling .pio\build\d1_mini\libaec\AHRS\Madgwick.cpp.o
Compiling .pio\build\d1_mini\libaec\AHRS\Mahony.cpp.o
Archiving .pio\build\d1_mini\libaec\libAHRS.a
Compiling .pio\build\d1_mini\libf83\EspSoftSerial\EspSoftSerial.cpp.o
Archiving .pio\build\d1_mini\libf83\libEspSoftSerial.a
Compiling .pio\build\d1_mini\libe6b\EspWire\EspWire.cpp.o
Compiling .pio\build\d1_mini\libe6b\EspWire\esp_twi.cpp.o
Archiving .pio\build\d1_mini\libe6b\libEspWire.a
Compiling .pio\build\d1_mini\libe23\Wire\Wire.cpp.o
Compiling .pio\build\d1_mini\libf2d\SPI\SPI.cpp.o
Archiving .pio\build\d1_mini\libe23\libWire.a
Compiling .pio\build\d1_mini\lib586\Kalman\Kalman.cpp.o
Archiving .pio\build\d1_mini\libf2d\libSPI.a
Archiving .pio\build\d1_mini\lib586\libKalman.a
Compiling .pio\build\d1_mini\lib627\EEPROM\EEPROM.cpp.o
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\BearSSLHelpers.cpp.o
Archiving .pio\build\d1_mini\lib627\libEEPROM.a
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
Compiling .pio\build\d1_mini\lib4d9\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
[... 144 lines cut ...]
Leaving...
Hard resetting via RTS pin...
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, 
nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
rl␀l�r␂$␒�n␌␌␌�␌l�␌b|��␂␇␒�r␒b�␌b␄�nn␂lnn␒b␌b␜p�$b␎lrlp�n␐␂␌␌�␌l␌�␂␌␌␌b␌n�n�␌�␄␌b��nn'l�␄l`␂�␒␒nn␌$`␂␎␂nr���n␌␌rr�`␂p�n0␂␌␌r�␜␜b ␄␂␌b␌n�n␌␄␌b��nn'␌␄l`␂�␒␒nn␌$`␂␎␂nr���n␌␌r␒l`␂␎r��n␌␌r␒l`␂�␂␌␌l$0�␌l`␂��n�`␂ I UART 0 0 1 115200 -1 -1 
 I UART 1 1 128 250000 -1 -1 
 I EEPROM loaded
 I I2C SETUP 1000 1
 I I2C DETECT MPU9250 0 
 I I2C DETECT MPU6050 1 
 I MOTOR CONF 5 0 480 1
 I MOTOR PIN 0 0
 I MOTOR PIN 1 14
 I MOTOR PIN 2 12
 I MOTOR PIN 3 15
 I GYRO INIT MPU6050 0 8000 2000 500
 I ACCEL INIT MPU6050 502 1990 1 
 I FUSION MADGWICK 50
 I RX PPM 13 1
 I GYRO BIAS -1.13 -1.51 0.86 

I also tried using the serial monitor on Arduino Ide to type status but there was no response back from the Wemos D1 mini serial

AutoPlantBali, 2023-06-24:
[image]

this response using putty

AutoPlantBali, 2023-06-24:
[image]

**stats** command

rtlopez, 2023-06-25:
Hi,

I created fix for serial on esp8266, but still can't connect from configurator 10.9. 10.8 works for me, but I'm not sure if the system matters here, I'm using Linux. If you want to test it, solution is in PR #33 (branch `serial-fix`)

Cheers,

AutoPlantBali, 2023-06-27:
hi rtlopez, thanks for fixing for connect betaflight configurator 10.8

i'am pull serial-fix, use windows 10, this same, configurator 10.9 not work but 10.8 works... esp8266 only gyro sensor mpu6050 with cpu_load average 85%

I don't know if it's still working normally after adding a motor and rx receiver with cpu_load above 85%

closed 2023-06-27 by AutoPlantBali

AutoPlantBali, 2023-06-28:
[image]

esp32 spi gyro MPU-9250 doesn't work, I've tried connecting to SPI0 or SPI1, if using I2C it works but the barometer doesn't work and load_cpu is up to 99% and sometimes it hangs on the uart communication

rtlopez, 2023-06-28:
Hi 

```
git checkout master
git pull
```
And try again, I just merged bunch of updates :)

AutoPlantBali, 2023-07-05:
result pull commit 2d713ef0fb35355ade33baf359c044538a3736c2

========== esp8266 use mpu6050 i2c =============
[image]

========== esp8266 use mpu9250 i2c =============
[image]

========== esp32 use mpu6050 i2c ===============
[image]

========== esp32 use mpu9250 i2c ===============
[image]

========== esp32 use mpu9250 spi0 ==============
[image]

===> spi is faster than i2c and load_cpu is smaller
===> brushed motor only works on esp8266 and esp32 doesn't work
===> esp8266 and esp32 connect and disconnect several times in betaflight sometimes it makes the settings return to default settings

PR #33 (2023-06-27): esp8266 serial fix
fixes #32