rtlopez/esp-fc #45: Esp32 not detecting gyro

Shally1986, 2023-07-26:
I have flashed the esp32 firmware to esp32 nodemcu and wired up mpy6050 as per the image on the docs . But in betaflight there is no gyro . 

I think the image is for spi connection and mpu6050 is i2c . Where do I connect mpu6050 on esp32 ?

AutoPlantBali, 2023-07-27:
mpu6050 use i2c only..
esp32   | mpu6050
D22      | SCL
D21      | SDA

Shally1986, 2023-07-27:
Thanks it works . I'm using esp 32 wroom . Pid loop is fixed at 1K how to make it 8K

AutoPlantBali, 2023-07-27:
use mpu6500 or mpu9250 support spi

Shally1986, 2023-07-27:
Thanks for the information .

Awesome project . I'm trying to build a brushed flight controller .

Shally1986, 2023-07-27:
Issues closed

closed 2023-07-27 by Shally1986

rtlopez, 2023-07-27:
Hi @Shally1986 @AutoPlantBali,

actually not quite, I checked it and I2C does not work on esp32. So I prepared fix and just merged to master branch. If you want to test it, flash device directly from this branch, I'm going to prepare new release soon.

MPU6050 works only on I2C bus.

Also keep in mind that wiring for esp32 might not be up to date, werify your pinout with CLI output command `get pin`, especially motor output. On ESP32 you can remap most pins according to your needs too.

Shally1986, 2023-07-27:
Post connecting mpu6050 to i2c I'm able to detect gyro and accelerometer in
betaflight .

How do I flash directly from branch ? Do i need to compile ? Could you help
me with the firmware or details on compilation and pin out for pi pico .

Thanks
Kushal Sahota

rtlopez, 2023-07-27:
@Shally1986 I forgot, you can try this build https://github.com/rtlopez/esp-fc/actions/runs/5679139675

RP2040 is experimental, and because of lack FPU, do not perform fast. I have not yet prepared the upload process without the development environment.

PR #47 (2023-07-27): I2C fixes
Fixes #45