rtlopez/esp-fc #46: wireless mode AP for softserial?

AutoPlantBali, 2023-07-27:
I try to edit this code
https://github.com/rtlopez/esp-fc/blob/add5b6ba346f0be9034b2920dda7488fdc7a1de8/lib/Espfc/src/Wireless.h#L33

with this code:

WiFi.persistent(false);
        
        if(_model.config.wireless.mode == WIRELESS_MODE_AP){
          WiFi.mode(WIFI_AP);
          status = WiFi.softAP(_model.config.wireless.ssidAp, _model.config.wireless.passAp);
          _server.begin(_model.config.wireless.port);
          _model.state.serial[SERIAL_SOFT_0].stream = &_adapter;
          _model.state.localIp = WiFi.softAPIP();
          _initialized = true;
          _model.logger.info().log(F("WIFI EV")).log(F("IP")).logln(WiFi.softAPIP());
        }else{
          WiFi.mode((WiFiMode_t)_model.config.wireless.mode);
          status = WiFi.begin(_model.config.wireless.ssid, _model.config.wireless.pass);
        }

on CLI set AP mode:
set wifi_mode AP
set wifi_ssid_ap autoplantbali
set wifi_pass_ap autoplantbali
set wifi_tcp_port 1111

setup pc windows/linux using virtual serial:
https://hackaday.io/page/1304-virtual-serial-port-tunnel-to-use-with-esp-link

betaflight configurator can connect with esp32 and android betaflight configurator with settings: tcp://192.168.4.1:1111

maybe plus 1 input button to turn off and on access point mode. so there is no need to connect using a USB cable and 3 serials can be fully used. 
I see the wifi code is connected to softserial, what impact will it have?

rtlopez, 2023-07-28:
Hi,

Thanks for your comment.

I implemented it as proof-of-concept, but testing shows, that it consumes a lot of processing time which block PID loop. It wasn't on my priority list later. Definately I do not recommend to have this option enabled while arming. There is no any signity check at the moment. Technically it should be enough for configuration purpose or maybe for OTA firmware upgrade.

You can also start AP on your phone and you should be able to check assigned IP on your android device instead starting AP on ESP. Access point mode provokes more communication, because the phone tries to get access to the internet through ESP, then it floods the device with lots of requests from its applications.

I don't get why you mention about virtual serial, while it is not necessary because BF configurator can connect to tcp directly. Last time I tested it on v10.7 IIRC.

By input button do you mean to use GPIO pin to enable or disable wireless? But as it requires additional hardware, I'd consider to control this function with spare RC channel.

Exposing as soft serial allows to control what functions are enabled on this data stream. I do not see any negative impact here.

According to code you presented, it woud be better to move some parts to wifi event callback and handle `SYSTEM_EVENT_AP_STAIPASSIGNED` and `SYSTEM_EVENT_AP_STOP`, especially:
```cpp
_model.state.serial[SERIAL_SOFT_0].stream = &_adapter;
_model.state.localIp = WiFi.softAPIP();
```

If I found more time I'll play with it bit more.

AutoPlantBali, 2023-07-29:
Hi,
Thank you for replying and explaining in detail for me.

I have tried STA and AP modes, in STA mode there is more delay when connected to the betaflight configurator. if AP mode blocks PID loop maybe after arming auto turns it off.

after I try it can directly connect to tcp directly. thanks for the info, I usually use virtual serial to program Arduino via esp bridge.

wifi event callback
when connected to AP:
I WIFI EV SYSTEM_EVENT_STA_WPS_ER_PIN
I WIFI EV SYSTEM_EVENT_AP_STOP

when disconnected to AP:
I WIFI EV SYSTEM_EVENT_AP_START

this is the response that appears when rebooting via cli

closed 2023-07-29 by AutoPlantBali