rtlopez/esp-fc #48: Brushed motor test

Shally1986, 2023-07-27:
On esp32 wroom board , mpu 6050 on i2c , elrs . Gyro accelerometer and elrs receiver all detected on betaflight . 

I hav connected pin out 4 to brushed motor driver (fet ) . But on motor test no motor spool up . I hav connected battery to the motor fet and pin out 4 to pwm input . 

Is there any specific setting for brushed ?

rtlopez, 2023-07-27:
Hi, there is new release https://github.com/rtlopez/esp-fc/releases/tag/v0.2.0-alpha2 where issue with brushed PWM should be fixed. Let me know if it helped.

Shally1986, 2023-07-28:
Hello , 

It didn't help but I'm not sure if it's software or hardware problem . I was able to run the motors on Arduino code using pwm . I will recheck again if the fet got fried or something . Do you have any schematic for brushed if you have tried ?

AutoPlantBali, 2023-07-29:
via cli in betaflight configurator:
use this command
# get pin_output
set pin_output_0 27
set pin_output_1 25
set pin_output_2 4
set pin_output_3 12
set pin_output_4 -1
set pin_output_5 -1
set pin_output_6 -1
set pin_output_7 -1

please check the vgs of the mosfet you are using.
I tried on pin 4 and it runs normally
[image]

[image]

mosfet brushed schematic:
[image]

Shally1986, 2023-07-29:
I will check and get back . Another stupid question . It will work with fets only or can use h bridge drivers ?

AutoPlantBali, 2023-08-07:
why use h bridge?, you need 2 pins to control 1 motor

Shally1986, 2023-08-07:
I think I do not have the correct mosfets and ordering them would take a
while . I have couple of lightweight tb6612fng motors drivers ( h bridge )
. As per my understanding h bridge is needed , if motors need to run both
direction . Most brushed flight controller use MOSFET but there were a few
which used h bridge and also supported turtle mode .

closed 2023-10-11 by rtlopez