rtlopez/esp-fc #62: Not arm my quad

skimans, 2023-08-19:
Hi
i have install last version of firmware for my ESP32 but when i connect battery not arm the quad.
I use MPU6050 and my radio is Flysky I6.

how i can solve ?

thanks
[image]
[image]
[image]
[image]
[image]

rtlopez, 2023-08-19:
Hi @skimans, 

It seems that you gyro is very noisy and callibration not finishing. Ensure your model is not moving when you connecting it. Consider mounting it with using adhesive tape and solder with flex cables instead direct soldering. If it won't help, you should look for replacement.

skimans, 2023-08-19:
Hi
I have solved.
Was a problem on my gyro/acc.
I have used GY87 but not work becouse i have used 3,3 volt.
Now i have used GY521...normal MPU6050 and now work fine

One question . It's posible have a version firmware for Inav ?

Thanks much.
And my complimentes for this your wonderful project.

Stefano

rtlopez, 2023-08-19:
What do you mean by 
> firmware version for Inav?

This firmware is neither Betaflight nor Inav. It is only partialy compatible to work with Betaflight-configurator. Not all features are available here.

closed 2023-08-29 by rtlopez