rtlopez/esp-fc #69: ESP32-C3 support?

bsvdoom, 2023-08-25:
I started to make a new target for the ESP32-C3.
What do you think, can it be done within a reasonable time for a noob like me, or i should give up, because it is a too complex task bc of the different architecture?

This is where i am at now:

`02:18:06.995 -> ESP-ROM:esp32c3-api1-20210207
02:18:07.027 -> Build:Feb  7 2021
02:18:07.027 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
02:18:07.027 -> SPIWP:0xee
02:18:07.027 -> mode:DIO, clock div:1
02:18:07.027 -> load:0x3fcd5810,len:0x438
02:18:07.027 -> load:0x403cc710,len:0x91c
02:18:07.027 -> load:0x403ce710,len:0x25b0
02:18:07.027 -> entry 0x403cc710
02:18:07.124 -> [   114][E][esp32-hal-mis�� I UART 0 0 1 115200 21 20 
 
02:18:07.124 ->  I F_CPU 160000000 
 
02:18:07.124 ->  E EEPROM wrong magic 
 
02:18:07.124 ->  I SPI SETUP 6 5 7 1 
 
02:18:07.124 ->  I I2C SETUP 9 8 1000 1 
 
02:18:07.124 ->  I SPI DETECT MPU9250 10 0 
 
02:18:07.124 ->  I SPI DETECT ICM20602 10 0 
 
02:18:07.124 ->  I SPI DETECT LSM6DSO 10 0 
[   134][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
02:18:07.155 ->  
02:18:07.155 ->  I I2C DETECT [   141][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
02:18:07.155 -> MPU9250 0 
 
02:18:07.155 ->  I I2C DETECT ICM20602 0[   149][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
02:18:07.155 ->  
 
02:18:07.155 ->  I I2C DETECT MPU6050 0 
 [   159][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
02:18:07.155 -> 
02:18:07.155 ->  I I2C DETECT LSM6DSO 0 
 
02:18:07.187 ->  I MOTOR CONF 9 0 480 0 
 
02:18:07.187 ->  I MOTOR PIN 0 2 
 
02:18:07.187 ->  I MOTOR PIN 1 4 
 E (188) rmt: rmt_set_gpio(530): RMT CHANNEL ERR
02:18:07.187 -> E (188) rmt: rmt_config(686): set gpio for RMT driver failed
02:18:07.187 -> 
02:18:07.187 ->  I MOTOR PIN 2 E (198) rmt: rmt_set_gpio(530): RMT CHANNEL ERR
02:18:07.187 -> E (198) rmt: rmt_config(686): set gpio for RMT driver failed
02:18:07.187 -> 18 
 
02:18:07.187 ->  I MOTOR PIN 3 19 
 
02:18:07.187 ->  I FUSION MAHONY 50 
 `

AutoPlantBali, 2023-08-25:
i'm not sure it will run well with limited number of IO pins and only have 1 core

bsvdoom, 2023-08-25:
the esp8266 has less pins and 1 core as well, and it runs on that too..

bsvdoom, 2023-08-25:
Some more work, and all options seems to work in betaflight, but:
-Gyro slowly rotates to the right at 5dec/sec
-Motor outputs does not seem to work correctly
-Cannot set the PID cycle above 0.5kHz
-If I enable the baro, it freezes on the next boot after the message "I I2C DETECT MPU6050"

rtlopez, 2023-08-25:
@bsvdoom Thanks, that looks pretty decent. If you could do a PR, I'd put some comments. Nevertheless, the fact is that in the case of the C3 we have fewer IO pins available than in the ESP8266 IIRC.

bsvdoom, 2023-08-27:
@rtlopez As far as I know ESP8266 has 17 (ESP12-F has 9), and ESP32-C3 has 21 available IO pins.
I made the PR #70, so I close this issue now, thanks!

closed 2023-08-27 by bsvdoom