rtlopez/esp-fc #72: Gyro issue? (GY-521)

bsvdoom, 2023-08-29:


rtlopez, 2023-08-29:
Hi @bsvdoom 

go to the CLI, type `reboot` and wait untli you see a line `I GYRO CAL...`. Ensure, that gyro is not moving at that time. then place result here.

You can try to slow I2C bus with commands
```
set i2c_speed 700
save
reboot
```
you can try eventually 500 or 400.

bsvdoom, 2023-08-29:
```
15:42:14.638 -> # reboot 
15:42:14.638 -> 
15:42:14.638 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,7)
15:42:14.638 -> 
15:42:14.638 -> load 0x4010f000, len 3424, room 16 
15:42:14.638 -> tail 0
15:42:14.638 -> chksum 0x2e
15:42:14.638 -> load 0x3fff20b8, len 40, room 8 
15:42:14.638 -> tail 0
15:42:14.638 -> chksum 0x2b
15:42:14.638 -> csum 0x2b
15:42:14.638 -> v00061dc0
15:42:14.638 -> ~ld
15:42:14.702 -> ��n�r��n|�l�l`bbrl�nB�nl`�rl�l� I UART 0 0 1 115200 -1 -1 
 
15:42:14.702 ->  I F_CPU 160000000 
 
15:42:14.702 ->  I EEPROM loaded 
 
15:42:14.702 ->  I I2C SETUP 4 5 700 1 
 
15:42:14.702 ->  I I2C DETECT MPU9250 0 
 
15:42:14.702 ->  I I2C DETECT ICM20602 0 
 
15:42:14.831 ->  I I2C DETECT MPU6050 1 
 
15:42:14.831 ->  I I2C DETECT HMC5883 0 
 
15:42:14.831 ->  I I2C DETECT BMP280 0 
 
15:42:14.831 ->  I I2C DETECT BMP085 0 
 
15:42:14.831 ->  I MOTOR CONF 1 0 480 1 
 
15:42:14.831 ->  I MOTOR PIN 0 0 
 
15:42:14.831 ->  I MOTOR PIN 1 14 
 
15:42:14.831 ->  I MOTOR PIN 2 12 
 
15:42:14.831 ->  I MOTOR PIN 3 15 
 
15:42:14.831 ->  I GYRO INIT MPU6050 0 8000 1000 1000 
 
15:42:14.831 ->  I ACCEL INIT MPU6050 501 1995 1 
 
15:42:14.831 ->  I FUSION MAHONY 50 
 
15:42:14.831 ->  I RX PPM 13 1 
 
15:42:43.069 ->  I GYRO BIAS -0.00 0.00 0.00 
```
 
After this if i connect to betaflight, it rotates only 1deg/sec, but gyro does not react if i rotate the gy-521.
After reconnecting the Wemos, the same happens as before.

OK, so it was the module, 
THIS ONE PRODUCES ERRORS:

[image]

THIS ONE WORKS FINE: (immediately biased, no yaw roll)

[image]

closed 2023-08-29 by bsvdoom

rtlopez, 2023-08-29:
Hi @bsvdoom 

```
15:42:14.831 ->  I RX PPM 13 1 
15:42:43.069 ->  I GYRO BIAS -0.00 0.00 0.00
```
definately there is something wrong with this gyro. This log shows, that calibration took about 30s, and returned 3x0.0. Normally it should take 2 seconds and bias should lay around +/- 5.0 for each axis.