rtlopez/esp-fc #76: Make this into a real FC

bsvdoom, 2023-09-01:
So I was thinking, if it is possible/favorable to make an esp32 chip based FC?
If its made with open source in mind, clones can exist, so it would not be a real "product", like ESP-CAM.

Pros:
-now iNav has an ESP32 target too
-it could be incredibly cheap opposed to stm32 based ~$60 FCs
-all chips can fit on one side
Cons:
-VTX will never happen on this chip

@rtlopez What are your thoughts on this?
Thanks!

skimans, 2023-09-02:
@bsvdoom Vtx can connected in Softserial port and GPS in RX1 TX1.
With Betaflight there is also GPS_rescue for recover the quad.
Sure ..Inav is a best solution for long and stabilized range.

rtlopez, 2023-09-02:
Hi @bsvdoom, the market is already saturated with similar solutions and it will be extremally difficult to succeed both in terms of cheapness and functionality. It all depends on how many people will be willing to invest in software development and board production. I don't think that boards would be significantly cheaper. This project is not oriented towards the commercial market, at least for now ;)

bsvdoom, 2023-09-03:
Hi @rtlopez, actually I did not found a lot of products for this use case, only the [ESP-Drone](https://www.aliexpress.com/item/1005005695006370.html).
I thought it can be ridiculously cheap as the ESP chip is well under $1, also the MPU6050 is $1 (especially in lager quantities), but I guess the PCB manufacturing process and so the price of it is the game breaker here. :(
Anyways, thanks for the response, if you change your mind in the future, I will be more than happy to help.

BTW I will release a PCB project so one can CNC and assemble at home easily an FC for this project, like this prototype:
[image]

Cheers!

closed 2023-09-03 by bsvdoom