rtlopez/esp-fc #79: Next steps for Esp-FC

skimans, 2023-09-04:
Hi friends,
my next  steps with Esp-fc that i want proceed is to use:

1) GPS Beitian 880 connected in UART2 or UART1 (mmmm....i have many dubts tht work fine on UART1.
2) install camera and VTX
3) in Softserial port put external OSD (similar MinimOSD or FrskyOSD).

Do you think that is posible all this ?

bsvdoom, 2023-09-04:
Hi, @skimans , i think its only doable on esp32 with 3 uarts (uart1 is for usb comm) uart2 to gps, 3 tx to vtx telemetry.
Gps is doable, ask @rtlopez when is he planning to imlement it, or you can try too.
I dont know telemetry is implemented or not yet.

skimans, 2023-09-04:
hi @bsvdoom Uart3 is already used for RX receiver.

bsvdoom, 2023-09-04:
> hi @bsvdoom Uart3 is already used for RX receiver.

Yes, and the RX is occupied, but the TX of uart3 can be used for telemetry IMO.

rtlopez, 2023-09-12:
Hi @skimans @bsvdoom I belive you know how open-source works :) I'll take into account your wishes when planinng new features, but I cannot promise anything. Nevertheless GPS and Navigation is very complex task, so do not expect it soon ;)

closed 2023-09-12 by rtlopez