rtlopez/esp-fc #80: MPU6500 not work

skimans, 2023-09-04:
hi friends
i have install MPU6500 using this pinout but not work:

[image]

i have connected with 3.3V (the same in Inav where work fine).

[image]

AutoPlantBali, 2023-09-04:
hi @skimans..
MOSI pin (IO23) on ESP32 connect to ADO pin on mpu6500

skimans, 2023-09-04:
> hi @skimans.. MOSI pin (IO23) on ESP32 connect to ADO pin on mpu6500

[image]

hi @AutoPlantBali 
i have connected now IO23 to ADO but not work.

AutoPlantBali, 2023-09-04:
oopss..sorry @skimans 
ESP32 PIN  --------> MPU6500 PIN
IO18           --------> SCL
IO19           --------> ADO
IO23           --------> SDA
IO5             --------> NCS

skimans, 2023-09-05:
@AutoPlantBali now work. Thanks much!!

closed 2023-09-09 by skimans