rtlopez/esp-fc #81: Baro BMP280 not recognized

skimans, 2023-09-06:
Hi friends
i have install baro BMP280 with connection 3.3v GND SCL(I22) SDA (I21) but not work.

[image]

rtlopez, 2023-09-06:
HI @skimans please specify "not works". How do you verify it? Double check connections. Go to the CLI, type `status` command, and then `reboot` and paste result here. Also you can flash board with arduino i2c scanner sketch to verify if device responds at `0x77` address. Keep in mind, that baro reading is only informational, can only be logged to blackbox, it is not used in any flight mode, so altitude hold is not possible yet.

skimans, 2023-09-07:
hi @rtlopez the external module baro BMP280 not recognized on the yellow panel icons .
Now i checked with Arduino and this is the result:

Scanning...
I2C device found at address 0x76  !
done

rtlopez, 2023-09-07:
Hi @skimans This module documentation says
> The I2C interface uses the following pins:
• SCK: serial clock (SCL)
• SDI: data (SDA)
• SDO: The I2C address decides the pin. If SDO connects to GND(0), the address is 0x76, if it connects to VDDIO(1), the address is 0x77. In this module, we have connected it to VDDIO, so the address should be 0x77.
• CSB: Must be connected to VDDIO to select I2C interface.

Is anything connected to SDO pin? This module supports SPI, you can share bus with gyro, only need separate CS line. Connect CSB pin to baro_cs pi, 13 py default.

skimans, 2023-09-07:
i have saw that if i use external module GY-87 ...that include also baro BMP085...the baro is recognized and work fine.
Single external BMP280 is not recognized.

skimans, 2023-09-07:
> Hi @skimans This module documentation says
> 
> > The I2C interface uses the following pins:
> > • SCK: serial clock (SCL)
> > • SDI: data (SDA)
> > • SDO: The I2C address decides the pin. If SDO connects to GND(0), the address is 0x76, if it connects to VDDIO(1), the address is 0x77. In this module, we have connected it to VDDIO, so the address should be 0x77.
> > • CSB: Must be connected to VDDIO to select I2C interface.
> 
> Is anything connected to SDO pin?
> 
> This module supports SPI, you can share bus with gyro, only need separate CS line. Connect CSB pin to baro_cs pi then, 13 by default.

the SDO is free. anything connected.
i dont understand what can i do for have 0x77 and connect it to I2c.....VDDIO ??

bsvdoom, 2023-09-07:
You should generally try and solve these problems first yourself, by googling and trying solutions.
The majority of problems you declare here has nothing to to with esp-fc, just wrong wiring and general electronic failures.
Do not get me wrong, we are here to help, but spamming the issues before you spend enough time to try to solve it yourself, wont help you in the long-terms.

Now they specifically told you what is the problem and what to do in this case (which you could've also lookup in forums, or in the BMP280 datasheet), that CSB and SD0 pins have to be connected to VCC (3.3V).
Here is a picture to make this clear:
[image]

skimans, 2023-09-07:
@bsvdoom i have already speand my time to search in Google. If i ask here it's becouse i dont have found other solutions.

Thanks you!

rtlopez, 2023-09-07:
@skimans take multimeter and ensure if there is no short-circuit to GND or other pin. If there isn't, you can safely connect it to 3.3V, otherwise you can burn LDO or even chips on boards. Normally there shoud be pull-up resistors on SDO and CSB, but it is not guaranted. Your BMP listen on 0x76 address just because SDO level is 0.

skimans, 2023-09-07:
@rtlopez now you must work for add GPS :) 
[image]
[image]

bsvdoom, 2023-09-07:
> @rtlopez now you must work for add GPS :) thanks much!!

I am sure if you donate to him he will work on it tomorrow =)

skimans, 2023-09-08:
> > @rtlopez now you must work for add GPS :) thanks much!!
> 
> I am sure if you donate to him he will work on it tomorrow =)

do you have already donate....right ?

bsvdoom, 2023-09-08:
> > > @rtlopez now you must work for add GPS :) thanks much!!
> > 
> > 
> > I am sure if you donate to him he will work on it tomorrow =)
> 
> do you have already donate....right ?

yes, code 🥇

skimans, 2023-09-08:
> > > > @rtlopez now you must work for add GPS :) thanks much!!
> > > 
> > > 
> > > I am sure if you donate to him he will work on it tomorrow =)
> > 
> > 
> > do you have already donate....right ?
> 
> yes, code :) and time to solve your issues :)

OK. Allora fatti i cazzi tuoi.
(use Goggle Translate for this..)

closed 2023-09-09 by skimans