rtlopez/esp-fc #82: Error to flash firmware

skimans, 2023-09-07:
hi
after using Arduino for check I2c address i have tried to reinstall firmware Alpha5, but when i flash i have this error message:

[image]

how i can solve it ?

rtlopez, 2023-09-07:
Hi @skimans 

1. this screenshot is too small, it would be good if you provide fragment with upload configuration
2. what have you done so far to resolve this issue?
3. have you tried change cable or reset flash 
4. have you read first google link https://www.google.com/search?q=md5+of+file+does+not+match+data+in+flash
5. do you have anything connected to gpio strapping pins 2 or 12?

skimans, 2023-09-07:
Hi @rtlopez  thanks for your patience.
I have read in the web also your link but i dont have found solution.
I have also changed the cable.
I think that i have bricked my ESP32 using Arduino.

Muchas gracias rtlopez. Creo que termina aca mi esperencia con ESP32.
Desculpame si te o molestado mucho a preguntarte muchas informaciones.
Sigue adelante asi con tu proyecto!
Felicidades!

rtlopez, 2023-09-07:
Hi @skimans No te rindas amigo y sigue adelante. Creo que lo lograrás :) Créeme, aprender también requiere paciencia :)

AutoPlantBali, 2023-09-08:
hi.. @skimans , follow this step:

=> connect esp32 and set bautrate 115200
[image]

=> set flash address and click program
[image]

=> finish
[image]

==> all file flash
[espfc.zip](https://github.com/rtlopez/esp-fc/files/12558669/espfc.zip)

skimans, 2023-09-08:
@AutoPlantBali thanks much! i have solved.
ESP32 recovered.

AutoPlantBali, 2023-09-08:
@skimans.... u're welcome... please close all issues that have been solved so that it's easier to handle

closed 2023-09-09 by skimans