rtlopez/esp-fc #86: v0.2.0-RC1 / SBUS is not working (only PPM works) / ESP32 (non mini)

ghehssalto, 2023-09-12:
Hello folks,

first of all that is an outstanding project thanks!

Currently I'm not able to use SBUS, however PPM is working, but I want to use SBUS :(.

My setup:
*Flysky FS-i6x+FSIA10B
*ESP32 (non mini)

rtlopez, 2023-09-12:
Hi @ghehssalto , can you show me the ports and receiver tabs? Which pin are you using to connect receiver? It must not be the same as ppm. You have to use serial rx pin of selected port. You can check pin assignments using cli command 'get pin'.

ghehssalto, 2023-09-12:
Hello now it's working THX!

But I have e new issue:

Every now and then it hangs completely and I have to reconnect and
[image]

then this message comes up
[image]

rtlopez, 2023-09-12:
@ghehssalto on which port did you set the serial rx option? is this problem on reveiver tab only or all? are you able to connect to CLI? if yes, type `reboot` and paste result here.

rtlopez, 2023-09-12:
@ghehssalto I forgot to mention, there are some quirks in connection to BF configurator. Usually it is impossible to connect right after flashing board. Powe cycle is required. Also if you enter CLI mode, you have to connect disconnect twice to leave. This issues shoud be eventually resolved in configurator, but it is not part of this project. Also similar issues might appear in connection of specific USB/UART chips or theirs drivers, especially when saving configuration that forces reboot.

In general I recommend:
1. disable auto-connect option in top of application window
2. disable autocomplete in configurator options

[image]

ghehssalto, 2023-09-13:
will try thx for your great support!