rtlopez/esp-fc #87: Problem with Supported devices via SPI

ghehssalto, 2023-09-13:
Hello,

currently the sensor MPU-92.65 is working in betafligth and the sensor MPU-9250/6500 is not working (Flag: NO_GYRO)

MPU-92.65 is working
[image]

MPU-9250/6500 is not working (Flag: NO_GYRO)
[image]

But I cant fix the problem

rtlopez, 2023-09-13:
Hi @ghehssalto, write here your connections between gyro and esp board.

ghehssalto, 2023-09-13:
yes sure:
[image]

ADO to pin 19
SCL to pin 18
SDA to pin 23
NCS to pin 5

rtlopez, 2023-09-13:
@ghehssalto It seems that your gyro module is based on MPU-6555. I've never meet such chip so far. I have to chek if it is comatible with MPU-6500 or MPU9250. In few days I can prepare special firmware to verify, but At the moment assume, it is not supported.

ghehssalto, 2023-09-13:
> @ghehssalto It seems that your gyro module is based on MPU-6555. I've never meet such chip so far. I have to chek if it is comatible with MPU-6500 or MPU9250. In few days I can prepare special firmware to verify, but At the moment assume, it is not supported.

Thank you very much for your research. How did you figure that out that it is based on the MPU-6555. Do you have a Link to taht page?

I bought that one MPU-9250/6500:
https://de.aliexpress.com/item/4001201664263.html?spm=a2g0n.order_detail.order_detail_item.3.44e463682DCVEh&gatewayAdapt=glo2deu
(Note Pic is not matching the pic from above to the MPU-9250/6500 sensor.

rtlopez, 2023-09-13:
@ghehssalto I googled symbols on a chip. It is probably compatible, but retuns diferrent id so that it is not detected. With chinese counterfeit you never know what you get. I need to verify datasheet first.

skimans, 2023-09-13:
hi @ghehssalto this is MPU6500.
Many modules from Cina of this are MPU6500. 
Few or none are MPU9250.

ghehssalto, 2023-09-13:
> @ghehssalto I googled symbols on a chip. It is probably compatible, but retuns diferrent id so that it is not detected. With chinese counterfeit you never know what you get. I need to verify datasheet first.

okay, are you looking @ the datasheet for mpu 6555?

ghehssalto, 2023-09-13:
> hi @ghehssalto this is MPU6500. Many modules from Cina of this are MPU6500. Few or none are MPU9250.

how do you know that or do u just assume? And If so why I have to do a workaround (see my first message above) to get them work? Because of de decide id?

AutoPlantBali, 2023-09-13:
hi @ghehssalto ...
I just added the mpu6555 ID to this firmware, you can try flashing it with this firmware
[firmware.zip](https://github.com/rtlopez/esp-fc/files/12601651/firmware.zip)

ghehssalto, 2023-09-13:
@AutoPlantBali 
Thank you, but the connection is not made properly. It keeps like that:
[image]

AutoPlantBali, 2023-09-13:
@ghehssalto  Is there a serial monitor running in another application besides betaflight configurator?

ghehssalto, 2023-09-13:
@AutoPlantBali no, did you modify the most recent firmware version? Or an older one?

AutoPlantBali, 2023-09-13:
@ghehssalto I used v0.2.0-alpha5 and modified it, adding ID mpu6555 (0x7C) to the 6500 library. The chip can be read but can't use the library on the 6500 so it crashes when connected in betaflight

[image]

ghehssalto, 2023-09-14:
Could we try the same approach using v0.2.0-RC1?

AutoPlantBali, 2023-09-14:
@ghehssalto  I have made it with base v0.2.0-RC1.. flash with this firmware, if you still can't connect to the betaflight configurator, try opening the serial monitor (115200) and press the reset button, copy the log that appears.

[firmware.zip](https://github.com/rtlopez/esp-fc/files/12606929/firmware.zip)

ghehssalto, 2023-09-14:
@AutoPlantBali still not connecting to betaflight:

Serial monitor output:
[image]

AutoPlantBali, 2023-09-14:
@ghehssalto problem flash firmware, esp32 boot loop
same isssue
https://github.com/rtlopez/esp-fc/issues/82#issuecomment-1711475791

[image]
[fix-imu.zip](https://github.com/rtlopez/esp-fc/files/12607827/fix-imu.zip)

AutoPlantBali, 2023-09-14:
using all file in fix-imu.zip

skimans, 2023-09-14:
0x00 when flash firmware

ghehssalto, 2023-09-14:
> @AutoPlantBali still not connecting to betaflight:
> 
> Serial monitor output: [image]

Thank you, now it is connecting to betaflight but still no gyro is showing up. Same behavior as before. 
"if I plug in the sensor MPU-92.65 open betaflight and connect & then unplug the sensor an plug in the
sensor MPU-9250/6500 gyro is workings."

ghehssalto, 2023-09-14:
> using all file in fix-imu.zip

Thank you, now it is connecting to betaflight but still no gyro is showing up. Same behavior as before. 
"if I plug in the sensor MPU-92.65 open betaflight and connect & then unplug the sensor an plug in the
sensor MPU-9250/6500 gyro is workings."

AutoPlantBali, 2023-09-14:
Go to CLI, type reboot then enter... copy all the logs that appear here

ghehssalto, 2023-09-14:
Getting that log:

//# reboot 
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
 I UART 0 0 1 115200 1 3 
 
 I UART 1 1 1 115200 33 32 
 
 I UART 2 2 64 100000 17 16 
 
 I F_CPU 240000000 
 
 E EEPROM wrong magic 
 
 I SPI SETUP 18 23 19 1 
 
 I I2C SETUP 21 22 1000 1 
 
 I SPI DETECT MPU9250 5 0 
 
 I SPI DETECT MPU6500 5 0 
 
 IMU MPU6555 
 I SPI DETECT MPU6555 5 0 
 
 I SPI DETECT ICM20602 5 0 
 
 I SPI DETECT LSM6DSO 5 0 
 
 I I2C DETECT MPU9250 0 
 
 I I2C DETECT MPU6500 0 
 
 I I2C DETECT ICM20602 0 
 
 I I2C DETECT MPU6050 0 
 
 IMU MPU6555 
 I I2C DETECT MPU6555 0 
 
 I I2C DETECT LSM6DSO 0 
 
 I MOTOR CONF 9 0 480 0 
 
 I MOTOR PIN 0 27 
 
 I MOTOR PIN 1 25 
 
 I MOTOR PIN 2 4 
 
 I MOTOR PIN 3 12 
 
 I MOTOR PIN 4 -1 
 
 I MOTOR PIN 5 -1 
 
 I MOTOR PIN 6 -1 
 
 I MOTOR PIN 7 -1 
 
 I FUSION MAHONY 50 
 
 I RX SBUS

AutoPlantBali, 2023-09-14:
I SPI DETECT MPU6555 5 0
I I2C DETECT MPU6555 0

not detected in spi and i2c for mpu6555, has this sensor ever been used and is working normally? Or is this the first time you use it when you buy it?

ghehssalto, 2023-09-14:
Yes it is the first time, I use it. If I read the register 117 (who_am_i) i receive the following integer values respectively.

sensor MPU-92.65 -> 112 (as it should be)

sensor MPU-9250/6500 -> 117

Maybe it's helpful.

AutoPlantBali, 2023-09-14:
@ghehssalto MPU-92.65 response 112 this is MPU6500
MPU-9250/6500 response 117 this is Unknown or fake Invensense IMU, use 'IMU_Generic'

reference:
https://github.com/AutoPlantBali/FastIMU/blob/17ade12e96c771c1a5fccb142871a039a173061f/examples/IMUIdentifier/IMUIdentifier.ino#L40C29-L40C78

AutoPlantBali, 2023-09-14:
@ghehssalto do you want to try again, I made a response ID set to 117..

--> set flash address 0x10000
[image]

-->use this file:
[firmware.zip](https://github.com/rtlopez/esp-fc/files/12613970/firmware.zip)

Go to CLI, type reboot then enter... copy all the logs that appear here

ghehssalto, 2023-09-14:
@AutoPlantBali thx.
Now it is working! gyro is reachable!
Did u just changed the Device ID? Could you pls send my the source files of that specifc firmware pls?

current log:

//# reboot 
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
 I UART 0 0 1 115200 1 3 
 
 I UART 1 1 1 115200 33 32 
 
 I UART 2 2 64 100000 17 16 
 
 I F_CPU 240000000 
 
 I EEPROM loaded 
 
 I SPI SETUP 18 23 19 1 
 
 I I2C SETUP 21 22 1000 1 
 
 I SPI DETECT MPU9250 5 0 
 
 I SPI DETECT MPU6500 5 0 
 
 IMU MPU6555 
 POWER UP  1 
 SET CLOCK 
 SET CLOCK  1 
 SET SLEEP 
 SET SLEEP  1 
 I SPI DETECT MPU6555 5 1 
 
 SET DLPF MODE 
 SET DLPF MODE  1 
 GET RATE 
 I MOTOR CONF 9 0 480 0 
 
 I MOTOR PIN 0 27 
 
 I MOTOR PIN 1 25 
 
 I MOTOR PIN 2 4 
 
 I MOTOR PIN 3 12 
 
 I MOTOR PIN 4 -1 
 
 I MOTOR PIN 5 -1 
 
 I MOTOR PIN 6 -1 
 
 I MOTOR PIN 7 -1 
 
 SET DLPF MODE 
 SET DLPF MODE  1 
 GET RATE 
 SET RATE 
 GET RATE 
 SET RATE  1 
 SET FULL GYRO RANGE 
 SET FULL GYRO RANGE  1 
 I GYRO INIT MPU6555 0 GET RATE 
 8000 8000 125 
 
 SET FULL ACCEL RANGE 
 SET FULL ACCEL RANGE  1 
 I ACCEL INIT MPU6555 501 1995 1 
 
 I FUSION MAHONY 50 
 
 I RX SBUS 
 
 I GYRO BIAS 3.57 1.02 -0.18

AutoPlantBali, 2023-09-14:
@ghehssalto some of the articles I read used the mpu9250 library, do you want to try again... this time I used the mpu9250 library... follow the method above and use the files I included

[base on mpu9250.zip](https://github.com/rtlopez/esp-fc/files/12614285/base.on.mpu9250.zip)

Which of the two firmwares I made works well? I'll pull it over here later

same isssue:
https://github.com/kriswiner/MPU9250/issues/486

ghehssalto, 2023-09-14:
> @ghehssalto some of the articles I read used the mpu9250 library, do you want to try again... this time I used the mpu9250 library... follow the method above and use the files I included
> 
> [base on mpu9250.zip](https://github.com/rtlopez/esp-fc/files/12614285/base.on.mpu9250.zip)
> 
> Which of the two firmwares I made works well? I'll pull it over here later
> 
> same isssue: [kriswiner/MPU9250#486](https://github.com/kriswiner/MPU9250/issues/486)

works perfektly fine!!!
Dude you are awesome thx!

Could you pls share your modded src files?

AutoPlantBali, 2023-09-14:
@ghehssalto is the magnetometer working properly?

ghehssalto, 2023-09-15:
@AutoPlantBali no, but its fine because I bougth the MPU-9250/6500 as 6 axis gyro (description of the seller).
But the MPU-92.65 should be a 9-axis gyro but the magnetometer is also not showing up, do I have to change someting?
thx

ghehssalto, 2023-09-15:
@skimans both sensor do not show the magnetometer:

[image]

:(, is there anything else?

AutoPlantBali, 2023-09-15:
@ghehssalto MPU9250 with ID 0x75 (117) without magnetometer, only 6-axis

ghehssalto, 2023-09-15:
@AutoPlantBali yes also according to the seller description.

But what is with the MPU-92.65, that one should work but it's not :(

skimans, 2023-09-15:
GY-91 module have compass inside MPU9250

[image]

but is not good use this compass beocuse there are many noises in the fly.

ghehssalto, 2023-09-15:
@skimans thank you for the information. So you are alway switching off the magnetometer?

But the point is I bought the MPU-92.65 as a 9-axis gyro so I want to use alle features. If not possible I have to contact the supplier :(.

skimans, 2023-09-15:
@ghehssalto usually i use magnetometer of module GPS..for example Beitian 880.
Mag is important to use only with GPS for Return to Home of quad using Gps_Rescue function in Betaflight.
For now is not important and you can disactivate.

ghehssalto, 2023-09-15:
@skimans Thanks know I got to Idea.

But anway I want check that the magnetometer of MPU-92.65 is working

ghehssalto, 2023-09-15:
> @ghehssalto some of the articles I read used the mpu9250 library, do you want to try again... this time I used the mpu9250 library... follow the method above and use the files I included
>
> [base on [...]

Hello, I just checked today again and Heading is still increasing by +1 every 10 sec. using [base on mpu9250.zip](https://github.com/rtlopez/esp-fc/files/12614285/base.on.mpu9250.zip)

AutoPlantBali, 2023-09-15:
@ghehssalto disconnect and reconnect betaflight configurator...

ghehssalto, 2023-09-16:
@AutoPlantBali yes Heading keeps increasing by 1 deg/4 sec.

rtlopez, 2023-09-16:
@ghehssalto did you wait, until gyro calibration is finished? During this time gyro should not move. When finished you should see line "gyro bias ..." in terminal. Normaly it takes 2 seconds after powering board, but if gyro is poor or moving, it may never finish. Drift is symptom of lack callibration.

ghehssalto, 2023-09-16:
@rtlopez Thank you for your suggest. Yes I did it like that all the time, Heading is still increasing.

ghehssalto, 2023-09-16:
Could it be possible that neither my "MPU-92.65" nor my "MPU-9250/6500" has a magnetometer (tried via I2C & SPI BUS @ betaflight, with enabled magnetometer).

and device id 0x00 is responding random values
[image]

Read Reg. 0x00:
[image]

while Reg. 0x75 is returning 0x70 like it should be 
[image]

Read Reg. 0x75:
[image]

rtlopez, 2023-09-17:
@ghehssalto If it retuns 0x70 it is mpu6500, mpu9250 should return 0x71, so none of your boards have mag. In 9250 compas is connected through external i2c bus, so you cannot access it directly, but only in bypass mode. So by reading reg 0x00 you accesed gyro, not mag.

ghehssalto, 2023-09-17:
@rtlopez Okay thanks.

Mathew3000, 2023-10-05:
The chips seem to be rebranded / relabled MPU6555 (at least they seem to be the same functionally).

Mathew3000, 2023-10-05:
The Gyro is connected and seems to be working, though I did not yet test it in flight.

Mathew3000, 2023-10-05:
Sorry @AutoPlantBali already discovered that 👍🏻

Mathew3000, 2023-10-05:
I just continue this issue here, the gyto seems to work fine when connected to the PC, but as soon as I lift off, the quad starts spinning on the yaw axis quite fast

rtlopez, 2023-10-05:
Hi @Mathew3000 check these things first
1. ensure that model i setup tab moves exacly in the same way as your device
2. ensure that motors spin in direction presented in motor tab
3. ensure that motors are connected in specified order like in motors tab
4. verify setting "Motor direction is reversed" in motors tab

Mathew3000, 2023-10-05:
Hey, after calibration, the random drift got better. For your suggestions:
1. I did, the model does move as I would expect
3. They spin in correct order
and I guess it was 2. and 4. combined, when I started the motors did not spin as I expected so I set "Motor direction is reversed" BUT that did not reverse the motors. My mistake was that after that I changed the pins on the motor but forgot to unset the reverse setting. I will try to check that tomorrow morning.

rtlopez, 2023-10-05:
"Motor direction is reversed" desn't reverse the motors, it only tells firmware that motors are configured in that way, so can correctly command motors. Motors direction can only be set in esc configuration or rather by swapping motor wires (because there is no esc serial passthrough in this firmware).

Mathew3000, 2023-10-05:
I use Betaflight for 4 Years and have never needed it 🤦🏻‍♂️

skimans, 2023-10-06:
@Mathew3000 you can change direction of motors also in Bhleli Configurator of your Esc's.

Mathew3000, 2023-10-06:
So after resetting the `Motor direction is reversed` switch the copter flew. I managed to crash it after a barrel-roll, the controller seemed to have had a runaway and at the end of the roll it flipped away. I' will try the next attempts with my EFC32 when it arrives

caixiaohai0, 2023-10-18:
> GY-91 module have compass inside MPU9250
> 
> [image]
> 
> but is not good use this compass beocuse there are many noises in the fly.

Yours should be a refurbished gyroscope module. The refurbished chip does not have a magnetometer. Their prices are also different.

closed 2023-11-08 by rtlopez