rtlopez/esp-fc #89: Control esp-fc with WiFi

bshirazi, 2023-10-06:
Hi
Thank you for your works.
Is it possible to control multicopter via eso-fc with WiFi?
If yes is there android app or web based access?

Mathew3000, 2023-10-06:
Hi, do you mean control in flight or change settings over Wifi?

rtlopez, 2023-10-11:
Only configuration has been implemented, but rather as proof of concept. WiFi consumes to much CPU and is not reliable enough to be used to control such type of device.

closed 2023-10-11 by rtlopez

bshirazi, 2023-10-14:
> Only configuration has been implemented, but rather as proof of concept. WiFi consumes to much CPU and is not reliable enough to be used to control such type of device.

Thanks for reply.
May using an external Access Point reduce CPU load usage....