rtlopez/esp-fc #91: missing arduinofake.h library

khunglong207, 2023-10-13:
I cannot perform PIO test on your project. They are missing quite a few libraries in the Test folder, so can you fix it for me?
[image]

rtlopez, 2023-10-13:
Hi @khunglong207 unit test works only with native environment, and you need to have gcc insatelled in system. Try command `pio test -e native`

khunglong207, 2023-10-14:
Hi @rtlopez  I tried it but it didn't work, The screen appears as follows:
[image]
And then I tried testing but the system still reported missing arduinofake.h library. Can you send me the root directory containing this library because I am working on a project and developing the raspberry pi system to do image processing. So I really need your help.
[image]

rtlopez, 2023-10-14:
@khunglong207 You are still running test for `esp8266` environment. Select `native` in bottom bar.
[image]

khunglong207, 2023-10-15:
i am using esp8266 microcontroller, so i set 8266 environment.

khunglong207, 2023-10-15:
Can you make a video about setup, I'm quite a noob about this issue

Mathew3000, 2023-10-15:
I think the tests are meant to run with 'default' env, that does not affect your 'esp8266' code

khunglong207, 2023-10-15:
Het bro can u help ke connect from esp 8266 to beta fight

khunglong207, 2023-10-15:
@Mathew3000 
I TRIED trying to connect to beatfight but the system indicator is as shown below. How can I connect to betafight even though the baud rate I have set is 115200?
[image]

Mathew3000, 2023-10-15:
I use an esp32, but betaflight connects automatically as soon as I plug the esp into my computer, the message you see there indicates that the Configurator could not talk to / find a Betaflight controller on the port you selected

khunglong207, 2023-10-16:
@rtlopez , @Mathew3000 

i don't understand what is the problem here, i have an esp8266 circuit. I connected to the computer and then I imported the code into the device. Then when it comes to the test environment, the system reports missing libraries like arduinofake.h; fakeit.h; ….
Connecting to betafight software, the system indicates USB reception as ttyUSB0 but cannot connect as shown above. Help me

rtlopez, 2023-10-16:
Hi @khunglong207 could you elaborate what are you trying to achieve? 

To use this project you only need to run `build` and `flash` pio targets. Unit tests are designed to be able to run on local machine only and in the CI/CD pipline. On esp8266 `ArduinoFake` library is in conflict with `Arduino`. 

About connection issues, ensure that user you are logged in, is assigned to `dialout` group on Linux. Without that you will not be able to connect and flash device through `/dev/ttyUSB0`. This command might help `sudo usermod -a -G dialout $USER`, and then logout and login again.

khunglong207, 2023-10-16:
hello @rtlopez 
I am trying to make a flight controller on a quadrotor with an esp 8266 that uses an MPU6050 sensor; GPS
So first I tested the gryscope system by connecting the esp8266 and MPU6050 as shown in picture 
[image]
First I flash like the instructions you told me in github, I flash successfully as shown below (I used your latest version RC-02).
[image]
Then, I build code using by platformIO from VS code, next I Pio run. 
I opened betafight but it didn't work, even though I selected the correct tty/USB0 and the baudrate was 115200. I tried the code "sudo usermod -a -G dialout $USER" but it didn't work. What should I do next steps?
https://github.com/rtlopez/esp-fc/assets/142283805/2690d5db-d174-4c68-bce1-49d08042d6df

rtlopez, 2023-10-16:
hi @khunglong207 go to the options, and disable "advanced cli autocomplete". Let me know if it help, so I'll add this information to the redame file.

rtlopez, 2023-10-16:
@khunglong207 Also after flashing power cycle board by removing USB cable from port. For unknown reason betaflight-configurator is not able to connect right after flashing.

khunglong207, 2023-10-16:
The topology has been followed but it still doesn't work, can you teamview it with me?

rtlopez, 2023-10-16:
@khunglong207 try to connect using vscode terminal. If there are no messages, type `reboot` then later type `status` and paste result here.

khunglong207, 2023-10-16:
@rtlopez 

How to connect terminal vs code, I'm new so I don't really understand

khunglong207, 2023-10-16:
In the video, I connected using the platform in vs code so I'm pretty sure the code was run on esp8266 but even in betafight it still didn't work.

khunglong207, 2023-10-16:
@rtlopez 
i REBOOT AND STATUS IT
[image]

rtlopez, 2023-10-17:
@khunglong207 first picture shows that firmware works fine, gyro is correctly detected. I have no idea why configurator won't connect, it must be somethin with your system. I need some answers
1. Do you use more than one esp module? on first photo I see there is chip CH340, whilst on the screenshot above i see that that there is CP2102. Are you connecting to correct device? post here result of `lsusb` command, when board is connected.
2. are you connected in vscode terminal? You **cannot** use both at the same time.
3. Can you post here screenshot of the options tab?
4. what distribution and kernel version are you using?
5. try diferrent version of configurator 10.7 and 10.9 should work too.

khunglong207, 2023-10-17:
somehow i managed to connect to cp2102 drive with betafight but i can't connect to ch340 version i think because the drive version is different.
If I'm connected to betafight, I want to test run my motor, how can I do that? I'm in the motor section, but I don't know how to start it up. Can you help me?

Mathew3000, 2023-10-17:
Have you ever worked with Betaflight?
At this stage it should be basically just Betaflight standard stuff.
 - REMOVE YOUR PROPELLERS
 - connect power to your ESC
 - Toggle the "testmode switch"
 - SLOWLY move each motor slider up and down

[image]

caixiaohai0, 2023-10-18:
You should set the Flash Address to 0x0000 instead of 0x00

Mathew3000, 2023-10-18:
That should not make a difference, as both Values are refering to the start of the memory, same as 0x01 is the same as 0x0001

caixiaohai0, 2023-10-18:
> That should not make a difference, as both Values are refering to the start of the memory, same as 0x01 is the same as 0x0001

No, I also encountered this problem, this is my solution

Mathew3000, 2023-10-18:
I'm not doubting you, but that seems strange to me :D but I will remember :D

caixiaohai0, 2023-10-18:
> I'm not doubting you, but that seems strange to me :D but I will remember :D

Hahhh, it's not that you doubt it, but that I happen to have encountered this problem as well

khunglong207, 2023-10-18:
Thanks for all
but brushless motor not run this videohttps://github.com/rtlopez/esp-fc/assets/142283805/89dbacc8-b38f-42ae-bc98-e7069b94d40e

Mathew3000, 2023-10-18:
Did you select a motor protocol? and use the correct pins? because the beeping on your motor means the esc has not been initialised. This seems to be a PWM esc?

khunglong207, 2023-10-18:
yes this is brushless motor kv2212
esc XT60 ESC 40A
i connected correct Pin

Mathew3000, 2023-10-18:
Did you select the correct motor protocol? Because as long as the ESC is not initialised nothing will happen.

khunglong207, 2023-10-18:
i don't have a controller so i don't know the motor's protocol
can u help me

Mathew3000, 2023-10-18:
I mean, I can but don't you think it would be better to get a general understanding of what you are doing? I'm not sure if you should just use a FC project without knowledge of your stuff and just YOLO it... that seems dangerous. 
If I need to guess I'd say you need to use PWM. 
Also I would suggest you close this Issue, as this has nothing to do with the arduinofake library.

closed 2023-10-18 by khunglong207