rtlopez/esp-fc #92: Users/Community Discord

Mathew3000, 2023-10-18:
Hey, I noticed there is no Place for the Community under this Project to exchange, @rtlopez would you be willing to create a Discord or some sort? I think it could be a faster solution for Exchanges than GitHub issues ;)

rtlopez, 2023-10-18:
Hi guys, I see there is interest, give me a few days ;)

rtlopez, 2023-10-18:
here is the invitation https://discord.gg/jhyPPM5UEH, let's test ;)

rtlopez, 2023-10-20:
I'll keep it open until I update the documentation.

rtlopez, 2023-11-30:
@dahklfahdkghas What do you mean by 
""has expired"? this link is configured to never expire. Are you getting any error message?

dahklfahdkghas, 2023-11-30:
Yes, the discord app says it’s invalid or expired. But if you say it never expires I’ll try updating the app or something. I never use it normally.

bsvdoom, 2024-10-07:
> Hi all, thanks all for this great project, i wanted to start the esp-fc as a simple 4 channel rc receiver without gyro for fixed wing plane., where shall i start.

I think fixed wing is not supported yet.

If the discord server is ready, I think the invitation link can go to the README, and this ticket can be closed. Cheers