rtlopez/esp-fc #96: Crsf RX

ArshaanTech, 2023-11-04:
The crossfire RX receiver is receiving no input. 
The telemetry is connected and configured.
The receiver is working with other F4 or F7 controllers.
Any possible suggestions?

rtlopez, 2023-11-05:
@ArshaanTech The CRSF protocol is implemented to work with ELRS recivers. It seems to be based on some legacy version of CRSF. If you are able to enable it in receiver or force it to generate SBUS signal, the it might work.

ArshaanTech, 2023-11-07:
Oh okay i will try it... the elrs isn't working I'll try another receiver. I am using EP2 currently.

Mathew3000, 2023-11-07:
Are your pins correct? My setup works with ELRS

ArshaanTech, 2023-11-08:
my pins were all correct but i couldn't get it to work, now I'm using a ppm receiver which is working good.
Also, can the PWM output a 0-255 full like Arduino Uno or nano?

rtlopez, 2023-11-08:
@ArshaanTech 
1. show us your receiver pin connections here, like tx -> pin number, pin marked as tx in receiver must be connected to rx in fc
2. go to the cli and enter `get pin` ant then `get serial` and paste result here
3. in ports tab, you need to choose one uart as serial rx
4. Why would you like to see PWM output in 0-255 range? configurator is maintained by betaflight developers, I have no plan to fork this project.

khunglong207, 2023-11-09:
@rtlopez 

I am following the esp 8266 diagram, but I wonder how I want to connect to the receiver for the 8266? because I see the diagram has an rx ppm pin but I don't quite understand how it is connected. I hope you can post some pictures of them so I can understand better

rtlopez, 2023-11-09:
@khunglong207 esp8266 can only PPM, there is not enough UARTs in this chip.

closed 2023-11-15 by rtlopez