rtlopez/esp-fc #97: wobbly

ArshaanTech, 2023-11-11:
Hey! The flight controller is working! but there is a problem the drone keeps wobbling a lot which I think it shouldn't... anything I can do to help it?

Mathew3000, 2023-11-11:
Check the mounting of the IMU, this seems like a vibration issue to me. Otherwise check your pid tune

ArshaanTech, 2023-11-11:
The imu is mounted nicely with flex quick. pid is by default.

Mathew3000, 2023-11-11:
Depending on your quad you might need to tune the pid

ArshaanTech, 2023-11-11:
I don't actually know how to edit PID values... any suggestions?

ArshaanTech, 2023-11-11:
do you think i should rather use Double sided tape? or fevi quick is better

rtlopez, 2023-11-11:
@ArshaanTech We need more details about your build and configuration
1. list all used components and show photo of your build
2. go to the CLI tab and type `status` command and then `dump` and place result here
3. If you are using ESC that you asked in #94 then it is probably the reason, 60Hz is way below recommended rate, you need ESC, that supports OneShot125 protocol at least, and can be updated with 1000Hz.

ArshaanTech, 2023-11-11:
I changed the ESC im using a 4in BLHeli ESC with Dshot 600 firmware.
Im using ESP32 non mini, MPU6050 and a ppm receiver.
The frame is 3D printed custom.

Status: 
```# status 
ESPF ESP32 v0.2.0-RC2 e4ec8b3 Oct 11 2023 20:48:31 8.4.0 201103
STATUS: 
    cpu freq: 240 MHz
  gyro clock: 8000 Hz
   gyro rate: 2000 Hz
   loop rate: 2000 Hz
  mixer rate: 2000 Hz
  accel rate: 501 Hz
   baro rate: 0 Hz
    mag rate: 0 Hz

 gyro device: MPU6050/I2C
 baro device: NONE
  mag device: NONE
     rx rate: 43
     rx lpfs: 0.00, 0.14

 arming disabled: 65542
```
dump:
```
dump 
#ESPF ESP32 v0.2.0-RC2 e4ec8b3 Oct 11 2023 20:48:31 8.4.0 201103
set features 536870913
set debug_mode NONE
set debug_axis 1
set gyro_bus AUTO
set gyro_dev AUTO
set gyro_dlpf 256Hz
set gyro_align DEFAULT
set gyro_lpf_type PT2
set gyro_lpf_freq 100
set gyro_lpf2_type PT1
set gyro_lpf2_freq 213
set gyro_lpf3_type PT1
set gyro_lpf3_freq 150
set gyro_notch1_freq 0
set gyro_notch1_cutoff 0
set gyro_notch2_freq 0
set gyro_notch2_cutoff 0
set gyro_dyn_lpf_min 170
set gyro_dyn_lpf_max 425
set gyro_dyn_notch_q 300
set gyro_dyn_notch_count 4
set gyro_dyn_notch_min 80
set gyro_dyn_notch_max 400
set gyro_offset_x -156
set gyro_offset_y -97
set gyro_offset_z -19
set accel_bus AUTO
set accel_dev AUTO
set accel_lpf_type BIQUAD
set accel_lpf_freq 15
set accel_offset_x 292
set accel_offset_y -245
set accel_offset_z -134
set mag_bus AUTO
set mag_dev AUTO
set mag_align DEFAULT
set mag_filter_type BIQUAD
set mag_filter_lpf 10
set mag_offset_x 0
set mag_offset_y 0
set mag_offset_z 0
[... 216 lines cut ...]
set mix_40 0 0 0
set mix_41 0 0 0
set mix_42 0 0 0
set mix_43 0 0 0
set mix_44 0 0 0
set mix_45 0 0 0
set mix_46 0 0 0
set mix_47 0 0 0
set mix_48 0 0 0
set mix_49 0 0 0
set mix_50 0 0 0
set mix_51 0 0 0
set mix_52 0 0 0
set mix_53 0 0 0
set mix_54 0 0 0
set mix_55 0 0 0
set mix_56 0 0 0
set mix_57 0 0 0
set mix_58 0 0 0
set mix_59 0 0 0
set mix_60 0 0 0
set mix_61 0 0 0
set mix_62 0 0 0
set mix_63 0 0 0
```

rtlopez, 2023-11-11:
@ArshaanTech What motors, propellers and Lipo are you using, is it 3S or 4S? Are wobles fast or slow? Do they appear during manouvers or all the time? Are you able to record and share video?

At the beginning you can try
1. 1k pid loop if CPU load is above 50%
2. in pid tuning tab reduce master multiplier slider if wobles are fast or increase if they are slow

ArshaanTech, 2023-11-11:
i will share video by tomorrow. im using 1000kv motors and 6 inch propellers on a 3s wobbles are fast and they are all the time. The drone keeps wobbling and I have to get throttle way too high sometimes (100%) to control it. The CPU load is only 28% idle and reached max of 35 during caliberation.

rtlopez, 2023-11-11:
What motors are you using exactly? Are they 6S capable? If yes and you are using 3S, then you are getting only fraction of available power (about 25%). That would explain, that you have to raise throttle so high. Few more questions
1. what mode are you flying, acro or angle? does anything changes if you switch between them?
2. did you enabled air mode? 
If your motors and escs are 6S capable, and propeller size fits to your motor, it should be fine to test that configuration. As safety precondition, you may want to reduce pid gains with master multiplier on first flight.

I'm worry a bit that you are using 3D printed frame, as it might not be rigid enough. Stiffnes is very important. 

Also the best way to diagnose issues is to collect some logs with blacbox logger. This FC firmware operate with serial OpenLog or OpenLager.

ArshaanTech, 2023-11-12:
The motors aren't 6s capable, They're rated for 3-4s.
Im flying on angle I didn't go acro for now.
I think you're right the 3D printed frame could be a problem.

ArshaanTech, 2023-11-12:
> What motors are you using exactly? Are they 6S capable? If yes and you are using 3S, then you are getting only fraction of available power (about 25%). That would explain, that you have to raise throttle so high. Few more questions
> 1. what mode are you flying, acro or angle? does anything changes if you switch between them?
> 2. did you enabled air mode? 
> 
> If your motors and escs are 6S capable, and propeller size fits to your motor, it should be fine to test that configuration. As safety precondition, you may want to reduce pid gains with master multiplier on first flight. Defaults are for 4S/5inch setups.
> 
> I'm worry a bit that you are using 3D printed frame, as it might not be rigid enough. Stiffnes is very important. 
> 
> Also the best way to diagnose issues is to collect some logs with blacbox logger. This FC firmware operate with serial OpenLog or OpenLager.

Here is the image of the drone 
[image]

And also the video:

rtlopez, 2023-11-12:
@ArshaanTech for me it is slow woble/oscillation, I'm pretty sure, it is because of low pid gains and lack of air mode. So you can try
1. enable air mode in modes tab
2. In pid tuning tab move master gain slider to the right, by one step and try again

If you notice that it is better, but not enough, move slider next step. Repeat by one step until you feel locked in. If you notice, fast oscilations like shaking, and motor sound modulation, then step back, you hit the limit.

I don't think, that old esp32 module is the reason.

ArshaanTech, 2023-11-13:
Okay, I will change the frame and use a sturdy frame from the market. 
I will enable air mode and I will also try the PID slider.

ArshaanTech, 2023-11-15:
The same issue persists, The drone is still oscillating and is also drifting. I calibrated the accelerometer and gyro and the receiver values to 1500 uS. 
[image]
I ensured there are no moving parts in the drone. everything was stuck well with double-sided tape. Airmode was turned on and I increased PID values. 
Nothing works so far. 

https://github.com/rtlopez/esp-fc/assets/85013497/960be549-0ae5-4d93-b375-c3ef89ff3ea6

can there be any other cause for the issue?

Mathew3000, 2023-11-15:
This seems to be still an issue with your tune. Id say. That said, I have tried but never succeeded in flying the F450 frame with Betaflight(stm32)

Mathew3000, 2023-11-15:
Your propellers seem quite small to me compared to your framesize. But that might also just be from the images

ArshaanTech, 2023-11-15:
oh... okay so I guess this is just not the right flight controller for this quad?
The propellers are not the issue bcz they fit right. 
is it possible to share a footage of the drone you have built, like how stable it is during angle or acro? possible to show?

rtlopez, 2023-11-15:
@ArshaanTech My 5''/4S quad is rock solid https://www.youtube.com/watch?v=22mlmE2FoF0 
I have no experience with building such big quads, there is so many factors and without blackbox we can only guess, whats going on. I recommend to investing in this small device. Recently I've created setup instruction https://github.com/rtlopez/esp-fc/blob/master/docs/setup.md and there is mentioned how to connect sd logger.

The second video you've attached is exactly the same as first, was it intentional?

What exactly motors, propellers and batery are you using, maybe your quad is underpowered. Paste your new settings here.

ArshaanTech, 2023-11-15:
Okay i will use 10" props. and I'm sorry for the wrong attachment.

https://github.com/rtlopez/esp-fc/assets/85013497/49ab395d-bc95-437f-82a4-0ea3c1e187dc

The new settings are

1. 3s battery 60c 2200mAh
2. 2212 1000kV motors
3. BLHeli 30A 4 in 1 ESC (Dshot600)
4. 6" props - I'll use 10" from next time
5. F450 frame

ArshaanTech, 2023-11-16:
i will try with 1045, also is it better to use mpu6500 and 8khz instead of mpu6050, is there any noticeable difference?

rtlopez, 2023-11-16:
For experienced and demanding users, this may be noticeable. It provides better filtering, less delay, allows the use of higher pid gains and gives a better lock-in effect. But this also requires higher quality components and PID tuning skills. Personally, I recommend using the SPI interface if possible as I2C is about 10 times slower.

ArshaanTech, 2023-11-16:
Okay bro Thanks for the advice! I'll take it into consideration. Btw do you have any plans for working on VTX on screen display? 
I think it would be pretty nice to read analog battery voltage and display it on the FPV feed.

rtlopez, 2023-11-16:
Look for a Minim OSD device, it should work, via UART/MSP As far as I know.

ArshaanTech, 2023-11-19:
still the same problem. I'll just build a new quad when i have enough money.

ArshaanTech, 2023-11-19:
Do you intend on making newer firmware and solving the current betaflight problems? many things that I turn on just turn off automatically after a reboot, The qmc5883 sensor also not working. it's also better to add VTX support. I would say adding GPS and return to launch would be nice. i can help you with the testing if you want. Any plans?

rtlopez, 2023-11-20:
If you could choose only one function, what would it be?

ArshaanTech, 2023-11-22:
> If you could choose only one function, what would it be?

I would like it to be GPS. I think it's nice to have a return to launch at failsafe instead of the drone dropping from mid air. Atleast landing functionality should be enabled. Can you contact me on discord? I would love to help you out in any way I can, Although I'm not an expert at betaflight firmwares.

closed 2023-12-12 by rtlopez