rtlopez/esp-fc #108: support for esp32-s2 and esp32-s3

sysytwl, 2024-01-14:
I have an esp32-s3 board. tried. But it doesn't work. It may be a USB virtual serial port problem. Because I can feel the CPU temperature. Can anyone tell me where the serial port file is? Or how to modify it. I'm new to PlatformIO.

rtlopez, 2024-01-14:
s2,s3,c3 aren't released yet, but work is in progress. If you have enough development skills you can try to use this branch https://github.com/rtlopez/esp-fc/pull/106

closed 2024-01-14 by rtlopez

Boak29, 2024-01-29:
Hi mr, I have tried it and it was successful with my lolin s2 mini, thank you, but there are i2c error 4 errors, how can I fix it?

rtlopez, 2024-01-29:
Hi, this is not an issue, small numer of error occurs because FC try to detect devices that are not present. Problem is if the numer of errors increases.

Boak29, 2024-01-29:
many Thanks mr