rtlopez/esp-fc #120: BLHELiSuite Flash problem

sysytwl, 2024-05-18:
[image]

1. When I try to update the ESC firmware via the flight controller it fails. ESP32S3

[image]

2. All ESCs cannot be read. always

sysytwl, 2024-05-18:
[image]
can only configure the first esc.

rtlopez, 2024-05-20:
Can you provide information about
1. esp-fc version
2. esp chip version
3. blhelisuite version
4. esc firmware version
5. esc model
6. cli `dump` command result

sysytwl, 2024-05-22:
latest esp-fc master
esp32s3
16.7.14.9.0.2
g-h-20 ,16.9, https://github.com/JazzMaverick/BLHeli/tree/JazzMaverick-patch-1?tab=readme-ov-file
hakrc bls_20a 4 in 1 esc
dump:
[BTFL_cli_20240522_154855.txt](https://github.com/rtlopez/esp-fc/files/15405071/BTFL_cli_20240522_154855.txt)

sysytwl, 2024-05-24:
Using the same settings, I tried using the Bluejay Web Flash software. The first was successful.  But the rest failed.

bsvdoom, 2025-01-14:
I had the same issue, my only solution was disconnecting the ESC control pins from the FC, and configure the ESCs without the FC one by one.