rtlopez/esp-fc #121: Bdir-dshot 300 error 100%

sysytwl, 2024-05-18:
Hi, I'm using dshot300 with Bidir on. But I got 100% error in betaflight and RPM 0 all the time. I've tested it on a stm32f3 and it is working. hence the ESC is working correctly. is this normal?

rtlopez, 2024-05-20:
what type of esc are you using and what firmware is loaded to it?

sysytwl, 2024-05-22:
ESP32-S3
latest-esp-fc master, unchanged
G-h-20, 16,9, https://github.com/JazzMaverick/BLHeli/tree/JazzMaverick-patch-1?tab=readme-ov-file

rtlopez, 2024-05-23:
@sysytwl Why JazzMaverick? this firmware is not maintained for very long time. So far I've tested combination of ESP32 + Bluejay. For S3 I'm in process of completiing hardware ATM So I'll be able to verify in several days.

sysytwl, 2024-05-23:
ok, thx for the info. I'll check that.

sysytwl, 2024-05-24:
hi, I tried with Bluejay. It seems to have the same issue.

rtlopez, 2024-07-01:
@sysytwl in develop branch there are fixes for dshot-bidir, now rpm should be shown on ESP32-S3. If you are able to verify, that would be great.

sysytwl, 2024-07-09:
[image]
it's working. thx a lot.

PR #122 (2024-07-01): esp32-s3 bidir dshot support
fixes #121
closed 2024-07-09 by sysytwl