rtlopez/esp-fc #129: compiler optimization level

sysytwl, 2024-08-29:
with 
> build_unflags =
>  -Os

the size of the app rose from 88% to 95% on s3.

**plz also check esp32**

thank you

rtlopez, 2024-08-29:
Is this a problem now? The release build is with the -O2 flag. If it'll be neccesary, we can reconfigure the partition layout to get the missing space.

sysytwl, 2024-08-30:
Sorry, I didn't express myself clearly.
I mean if with unflags Os(small) the size of the app increased. I didn't remove the O2 in the [env]. which means the board configuration may contain an Os by default. Once you remove the Os flag by using the unflag function, the O2(performance) is applied, which causes the app size to increase a bit.
btw, I'm using 8M s3. I've changed the partition table. I have 6.7M space for the log. No issues so far.