rtlopez/esp-fc #131: Cant Flash esp32 s3 Zero Waveshare

saaraswata, 2024-09-14:
Flashing is successful but it shows an error in serial manager

entry 0x403c98d0
E (191) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x4037c592:0x3fceb180 0x40382571:0x3fceb1a0 0x4038843d:0x3fceb1c0 0x42027b4e:0x3fceb2f0 0x4037bf2b:0x3fceb320 0x403cd86b:0x3fceb350 0x403cdb2a:0x3fceb380 0x403c9925:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710

any fixes?

saaraswata, 2024-09-14:
Okay flashed it, had problems with partition

closed 2024-09-14 by saaraswata

colaco1123, 2025-01-29:
can you tell me how did you fix the issue?