rtlopez/esp-fc #133: flashing problem in esptool!

Zuhayer69, 2024-09-19:
I have a esp 32 dev module rev3 which i connected through usb and in esp tool i conected the board but I am am confused what file i should select for the firmware. Please tell me the firmware file name basically which file to choose to flash.

rtlopez, 2024-09-19:
espfc_v0.2.0-RC6_esp32_0x00.bin.zip, remeber to extract and follow guide https://github.com/rtlopez/esp-fc?tab=readme-ov-file#flashing

Zuhayer69, 2024-09-23:
> `espfc_v0.2.0-RC6_esp32_0x00.bin.zip`, remeber to extract and follow guide https://github.com/rtlopez/esp-fc?tab=readme-ov-file#flashing
Thanks for the reply I am going to test it as soon as possible and I will let you know

rtlopez, 2024-09-27:
esp32-c3 and esp32-s3 is not supported by this flasher. At the moment you need to use vscode with platformio extension.

Zuhayer69, 2024-09-27:
Will the schematics be same for the esp32 s3 module as the esp32 module? Or there is a difference ?

rtlopez, 2024-09-27:
no, pin assignment is diferrent. If you flash board, then go to the Cli and type `get pin` to see pin functions.

Zuhayer69, 2024-10-07:
the ADC pin for esp32 is two shown in the CLI one is ADC 0 and another is ADC 1 which one to use? please help

bsvdoom, 2024-10-07:
> the ADC pin for esp32 is two shown in the CLI one is ADC 0 and another is ADC 1 which one to use? please help

Try ADC 0

rtlopez, 2024-10-07:
ADC_0 is for voltage, ADC_1 is for current. Do not confuse it with internal `ADC` number of esp chip. Use only pins attached to `ADC1` of esp32.

closed 2024-11-23 by Zuhayer69