rtlopez/esp-fc #138: Custom Controller

KeithHanson, 2024-11-15:
Firstly, sir... Amazing work. I have both an esp32 and an esp32s3 wired up and registering in beta flight now 😍

https://bsky.app/profile/keithhanson.bsky.social/post/3layxvo6yos2l

I'd like to hook up a heltec Lora to this, but modify the code to support having the FC double as the remote control receiver. 

Firstly - is this even a good idea? I have yet to dive into the code, but I'm hoping I can use the multiple cores - one for FC work, one to receive packets.

My hope is to create a simple app on my steamdeck so I can use it as a ground station (FPV feed) and control (a simple pygame prototype for now that communicates with another heltec Lora via usb serial).

Secondly, if this is not a BAD idea and could be implemented, would you mind highlighting the areas of code to modify? 

I'm betting if I look around at how you've implemented the esp-now control, it'd be similar, but having your insight will save a lot of time :) 

Many thanks for this project! Can't wait to get my first esp32 controller drone in the air!

bsvdoom, 2025-01-14:
As far as I know, LoRa is only for very small packets and listening in the 99.99% of the time.
It also has very big latency, so it is not suitable to remote control anything real-time.

rtlopez, 2025-02-04:
@KeithHanson I admit that it would be interesting to have a steam deck as a controller with its FPV screen, if I got it correctly. But I don't know its capabilities at all, so it's hard for me to judge whether it's possible. I'm also not able to point out places for modifications, because first you would have to do some analysis and make some design decisions. But I hope that won't stop you from implementing it. From what I can say now, both cores are alredy utilized.

@bsvdoom Do you know ExpressLRS project? it's LoRa on SX1280 chip essencially.

closed 2025-02-04 by rtlopez

bsvdoom, 2025-02-06:
@rtlopez Ofc, I have a Radiomaster pocket, and I love it!
What I meant was that the SX1262 is designed specifically for LoRa (Chirp Spread Spectrum) and FSK (Frequency Shift Keying) modulation, and optimized for low data rates, and has a maximum data rate of around 27 kbps, which is far lower than what’s required for real-time drone control.
Also the round-trip latency is between 100 milliseconds and 500 milliseconds, or more.
The SX1262 is built for LoRa protocol and simple data transmission protocols (e.g., LoRaWAN), and it doesn’t support the necessary functionality for real-time control or frequency hopping.