rtlopez/esp-fc #140: Firmware File reference

auroraborealis207, 2024-12-18:
The Flashing section is denoting a .bin firmware file to upload ,I am unable to find any file related which is of the same format ,I also tried compiling to an image using docker but resulted in a .fzz file which failed to connect to beta flight , I am using the most lastest version of betaflight
[image]
[image]

auroraborealis207, 2024-12-18:
First Image is Beta Flight and Second from Flasher

auroraborealis207, 2024-12-18:
using ESP-Wroom-32

rtlopez, 2024-12-18:
On the releases page https://github.com/rtlopez/esp-fc/releases click **Assets**

auroraborealis207, 2024-12-19:
Thanks for the clarification!!!

closed 2024-12-19 by auroraborealis207