rtlopez/esp-fc #141: problem in connecting SBUS with esp32 s3 mini board (which pin to use!!??)

Zuhayer69, 2025-01-04:
I have a esp32 s3 mini board where i have flashed the firmware and now i want to connect the flysky receiver with SBUS protocol but which pin I should use I am confused and please tell what is the process to connect the receiver in beta flight. Here is the picture of the pins of the esp32 s3
[image]

rtlopez, 2025-01-04:
SBUS is serial protocol, that means you need to enable serial rx function in port tab on selected uart. Then select SBUS protocol in receiver tab. Once it is done you can connect receiver to pin mapped on `pin_serial_{0,1,2}_rx`.

Zuhayer69, 2025-01-05:
which one of the uart to use?
[image]

rtlopez, 2025-01-05:
use any free one convenient for you

Zuhayer69, 2025-01-07:
if i use uart 1 then i have to connect sbus pin to serial rx 0?

rtlopez, 2025-01-07:
yes

Zuhayer69, 2025-01-10:
i am facing problem with the battery. using a 3.7v battery i have connect 220k ohm and 1mega ohm resistor to make a voltage divider and connected it to the pin 1 but there is no voltage showing. do i have to do something in the power tab of beta flight?

rtlopez, 2025-01-10:
1. are you using 220k:1M or 1M:220k? Keep in mind, that fully charged battery has 4.2V. In this case I recommend 150k:220k divider (150k to battery, 220k to ground, or 15k:22k)
2. you must connect divider to pin assigned to `pin_input_adc_0` in cli
3. in power tab select "Onboard ADC" and tune scale to display real measurement (probably +/-10)

[image]

Zuhayer69, 2025-01-10:
i didnt knew the orientation mattered so im gonna fix that what about the calibration button?

Zuhayer69, 2025-01-10:
can you give me the equation of the divider and multiplier values

rtlopez, 2025-01-10:
if scale = 10, then for multiplier and divider you can use real resistor values. if scale = 100, then decrease multiplier 10 times or increase divider 10 times. Proposed values above should be good for start in your case. You can use calibration button.

closed 2025-02-04 by rtlopez