rtlopez/esp-fc #149: esp32s3 reset cycle

colaco1123, 2025-01-29:
I tried flashing an esp32s3 waveshare zero https://www.waveshare.com/wiki/ESP32-S3-Zero that i had lying around but it kept going on a boot cycle and kept resetting any solutions how to get it working on an s3?

colaco1123, 2025-02-04:
https://github.com/rtlopez/esp-fc/issues/131#issue-2526505580

here is the link to the same problem but he did not mention the solution

rtlopez, 2025-02-04:
1. Did you flash anything to this board before? 
2. Did you set "flash address" to `0x00` before programming in esptool?
3. Try to erase whole flash using online esptool https://espressif.github.io/esptool-js/ (consider reducing baudrate by half)
4. Try to flash example arduino sketch with usb enabled.

I have the same board and it works without issue. I don't think that issue is the same, this image contain partition layout in itself.

colaco1123, 2025-02-04:
i fixed it 

[image]

lol i ignored this instruction not " Add firmware file and set Flash Address to 0x00"

thanks for the help tho also amazing work

closed 2025-02-04 by colaco1123