rtlopez/esp-fc #151: Problem with ESP-NOW Configuration → SPI Rx (e.g. built-in Rx)

copilador, 2025-02-05:
Hello, I am trying to set up a drone with ESP-NOW (espfc_v0.2.1) enabled. If I understand correctly, I need to configure the position of SPI Rx (e.g., built-in Rx) in the receiver tab. However, after selecting SPI Rx (e.g., built-in Rx), saving, and restarting, I can no longer access the ESP32.

When I check the serial console, I see that it keeps rebooting with the following message:
Brownout detector was triggered

ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

I had to reprogram the ESP32, but when I tried again, it started rebooting continuously once more. I'm not sure what I'm doing wrong, or could the problem be due to using SPI to access the sensors? My configuration is:
set pin_input_rx 35    --->PPM ?? should be -1 ???
set pin_output_0 27    --->PWM -M1
set pin_output_1 25    --->PWM -M2
set pin_output_2 4     --->PWM -M3
set pin_output_3 26    --->PWM -M4
set pin_output_4 -1    
set pin_output_5 -1    
set pin_output_6 -1    
set pin_output_7 -1    
set pin_buzzer -1      
set pin_serial_0_tx 1  --->TX0
set pin_serial_0_rx 3  --->RX0
set pin_serial_1_tx 33--->TX1
set pin_serial_1_rx 32--->RX1
set pin_serial_2_tx 17--->TX2
set pin_serial_2_rx 16--->RX2
set pin_i2c_scl 22    --->SCL  Unused
set pin_i2c_sda 21    --->SDA  Unused
set pin_input_adc_0 36--->ADC VOLTAGE
set pin_input_adc_1 -1
set pin_spi_0_sck 18  --->SPI0 CLK
set pin_spi_0_mosi 23 --->SPI0 MOSI
set pin_spi_0_miso 19 --->SPI0 MISO
set pin_spi_cs_0 5    --->CS MPU-9250
set pin_spi_cs_1 13   --->CS BMP280
set pin_spi_cs_2 -1   
set pin_buzzer_invert 1

Thanks!

rtlopez, 2025-02-05:
Brownout means that your power supply or battery is not able to provide enough current so voltage drops below minimum.

copilador, 2025-02-06:
Thank you, you're right, it's a problem with the voltage regulator. When the radio part is activated, the consumption increases and the regulator I'm using, AP2112K-3.3, is not suitable. I'll have to find another solution...

closed 2025-02-06 by copilador

bsvdoom, 2025-02-15:
> Thank you, you're right, it's a problem with the voltage regulator. When the radio part is activated, the consumption increases and the regulator I'm using, AP2112K-3.3, is not suitable. I'll have to find another solution...

@copilador 
AP2112K-3.3 should handle 600mA, and ESP32 in active mode should only consume maximum 260mA.
However it can burst to 1A, so try adding a bigger cap before and after the AP2112.