rtlopez/esp-fc #153: receiver isnt getting detected

colaco1123, 2025-02-05:
I have an radiomaster r81 receiver with supports frskyd8 and gives a subs output i connected this signal pin to pin 17 on the esp32s3 supposedly the pin_serial_2_rx 17 but i don't seem to be able to get the receiver connected i think its something with regards to sbus inversion or so
i tried it with uart 1 too on pin 15 no luck 
can you please help me solve this issue thankyou

here is links to the receiver
https://quadkart.in/radiomaster-r81-8ch-frsky-d8-compatible-nano-receiver-w-sbus/

rtlopez, 2025-02-06:
1. did you bind receiver and transmitter?
2. attach screenshot of `ports` and `receiver` tabs

colaco1123, 2025-02-06:
yes the transmitter and receiver are connected and working fine 

[image]

[image]

colaco1123, 2025-02-06:
yes i have used this receiver before i used the sbus.h library 
and a definition like below (pin numbers are different)
bfs::SbusRx sbus_rx(&Serial2,11, 12,true);
bfs::SbusData data;

the wiring is just sbus to pin 17

colaco1123, 2025-02-06:
here check this under sbus it says "Some OpenLRS receivers produce a non-inverted SBUS signal. It is possible to switch SBUS inversion off using CLI command set sbus_inversion = OFF when using an F3 based flight controller."

but the esp_fc firmware doesnt have the set sbus_inversion parameter

https://betaflight.com/docs/development/rx

rtlopez, 2025-02-06:
if you choose UART3 then you must connect to pin assigned to `pin_serial_2_rx`, on esp32s3 by default is GPIO17 (in CLI they are 0indexed). On ESP32-S3-Zero this is an inner pad, so you might want to remap it to diferrent pin in CLI (see attachment) if you want to have it on outer pad, or use diferrent UART.

ExpressLRS has option to generate non-inverted SBUS, but it is not option in your case. Esp32 chip supports serial inversion and it is enabled fo SBUS, but firmware doesn't have any config to change it.

Does the code you attached works for you?

[image]

rtlopez, 2025-02-06:
do you have logic analyzer to capture signal trace?

colaco1123, 2025-02-06:
> if you choose UART3 then you must connect to pin assigned to `pin_serial_2_rx`, on esp32s3 by default is GPIO17 (in CLI they are 0-indexed). On ESP32-S3-Zero this is an inner pad, so you might want [...]

Yes the pin in connected to the inside pad infact before this i tried it with part 1 but it was the same 

Yes the code i attached works in getting sbus data onto the serial monitor in  Arduino

colaco1123, 2025-02-06:
> do you have logic analyzer to capture signal trace?
  
Yes I have one at my workplace i shall try to send you the waveform.

rtlopez, 2025-02-06:
On this picture I see that digital signal is transferred, but it would be good to see whole frame timings (frame time and space time) and idle level. Ideally at least 2 frames on a screen. Technically it should by about 3ms data frame, 6ms idle space, and 0 at idle. like this:

[image]

I've tested firmware v0.2.1 with Frsky XM+, and it works correctly.

Did you chang any pin assignment? go to the cli, type following commands, and paste results here (rx must be connected, ensure that is is powered by 5V)
```
get pin
status
logs
```

colaco1123, 2025-02-07:
ESPF ESP32S3 v0.2.1 120a1a6 Feb  4 2025 21:46:59 1.43 8.4.0 201103
STATUS: 
    cpu freq: 240 MHz
  gyro clock: 8000 Hz
   gyro rate: 2000 Hz
   loop rate: 2000 Hz
  mixer rate: 2000 Hz
  accel rate: 500 Hz
   baro rate: 54 Hz
    mag rate: 75 Hz

     devices: MPU6050/I2C, BMP085/I2C, QMC5883L/SLV
       input: 43 Hz, 0.00 Hz, 0.14
arming flags: FAILSAFE RX_FAILSAFE CLI MSP REBOOT_REQUIRED MOTOR_PROTOCOL
 rescue mode: 1
      uptime: 85.5

colaco1123, 2025-02-07:
well apparently after this i tried to flash it again and it works now thankyou so much for your help im an undergrad student and this is one of my few projects. i love your work thankyou so much for helping me.

[image]

closed 2025-02-07 by rtlopez