rtlopez/esp-fc #156: Issue when enabling motors in PWM mode?

copilador, 2025-02-15:
Hello, I’m having issues configuring motor control via PWM. I see that it is indicated as not recommended, but I believe it is supported? I’m using a nano drone with coreless DC motors (3.7V, 70,000 RPM) and ESP-NOW control mode. Everything seems to work correctly, except when I enable PWM mode with set output_motor_protocol PWM. After saving and restarting, the motors start running at high speed, and I can see a PWM signal on the GPIO outputs. Is PWM mode supported?
Thanks.

rtlopez, 2025-02-15:
PWM is for legacy ESC's, that don't support neither OneShot nor Dshot. Fo coreless motor driven by FETs, you probably want BRUSHED protocol.

copilador, 2025-02-15:
Thanks, I have switched to the BRUSHED protocol, but I get a message that I think means the outputs need to be assigned. I have this configuration for the 4 outputs:
set pin_output_0 27
set pin_output_1 25
set pin_output_2 4
set pin_output_3 26
set pin_output_4 -1
set pin_output_5 -1
set pin_output_6 -1
set pin_output_7 -1

[image]

[image]

copilador, 2025-02-15:
Hi, -in case it's useful- after doing some more testing, it works sometimes??, but most times it doesn't ??

[image]

rtlopez, 2025-02-15:
If you power from usb, test only one motor at a time. Usb do not have enough power, and board might reset.

copilador, 2025-02-15:
For the power supply, an adjustable laboratory power supply is being used, limiting the current to 2A and the voltage to 3.7V. The 3.7V is for the motors, and 3.3V (XC6220B331MR-G regulator) is for the GY-91 module and the ESP32. I am monitoring if there is a voltage drop, and it remains at 3.3V. The total current consumption with the motors at 100% is around 900mA. What I don’t quite understand is why I get the message:
 Mixer mode problem detected

The Quad X 1234 model needs 4 motor resources and the current configuration of the firmware provides 0 usable output(s) for the selected mode.

If using a custom mixer mode you need to define a custom mmix before changing mixer mode.

Please check your configuration and add the required motor resources.

And why is it sometimes solved just by changing an option in the Motors tab, without having to click Save and Restart? The most curious thing is that in the latest tests, it seems to work correctly.

If you think it's a hardware issue on my end, I will close the topic. Thanks again.

rtlopez, 2025-02-15:
`Quad X 1234` is quite unusual and probably not tested. I suppose this is the reason. use `Quad X` mixer.

copilador, 2025-02-15:
OK, I made the PCB with the Quad X 1234 layout, without thinking it could be an issue

[image]

closed 2025-02-15 by copilador