rtlopez/esp-fc #163: Using RX for GPS, TX for VTX IRC tramp

bsvdoom, 2025-03-20:
If I use CSRF the TX/RX lines will be occupied on that serial port, right?
So only UART3 remains, but I intend it to use for GPS in the future.
Or should use SOFTSERIAL instead for the IRC tramp when it will be implemented? - will it be? :)
Can the IRC tramp protocol even be implemented?
Which will be the optimal solution to this?
Thx

rtlopez, 2025-03-20:
@bsvdoom Basic TBS SmartAudio implementation is already in the master branch, see https://github.com/rtlopez/esp-fc/pull/159 . ELRS telementry also is there. If you have any idea to implement something, feel free to choose what fits your needs :)

rtlopez, 2025-03-20:
There are no ETA in this project :)

bsvdoom, 2025-03-21:
@rtlopez  
Ok, so this far as I could manage, help is appreciated: https://github.com/rtlopez/esp-fc/pull/159