rtlopez/esp-fc #170: No gyro on esp32se

anas-alali-24, 2025-05-14:
Hello

I have uploaded espfc_v0.2.1_esp32s3_0x00.bin  to my drone but i always get no gyro 
I tested the gyro with arduino ide  and I am geting the readings there
I used the cli to set the pins for i2c but nothing helps 
any Ideas?

[image]

Thanks

rtlopez, 2025-05-14:
1. what gyro module are you using?
2. describe wiring?
3. go to the cli and type commands: `logs`, `status`, `dump` and attach result here

anas-alali-24, 2025-05-14:
1- mpu6050
2- My own wiring ,but i tested the gyro and it is working
logs

[image]

status
[image]

dump
# dump 
defaults
set features 536870920
set debug_mode NONE
set debug_axis 1
set gyro_bus AUTO
set gyro_dev AUTO
set gyro_dlpf 256Hz
set gyro_align DEFAULT
set gyro_lpf_type PT1
set gyro_lpf_freq 100
set gyro_lpf2_type PT1
set gyro_lpf2_freq 213
set gyro_lpf3_type FO
set gyro_lpf3_freq 150
set gyro_notch1_freq 0
set gyro_notch1_cutoff 0
set gyro_notch2_freq 0
set gyro_notch2_cutoff 0
set gyro_dyn_lpf_min 170
set gyro_dyn_lpf_max 425
set gyro_dyn_notch_q 300
set gyro_dyn_notch_count 4
set gyro_dyn_notch_min 80
set gyro_dyn_notch_max 400
set gyro_rpm_harmonics 3
set gyro_rpm_q 500
set gyro_rpm_min_freq 100
set gyro_rpm_fade 30
set gyro_rpm_weight_1 100
set gyro_rpm_weight_2 100
set gyro_rpm_weight_3 100
set gyro_rpm_tlm_lpf_freq 150
set gyro_offset_x 0
set gyro_offset_y 0
set gyro_offset_z 0
set accel_bus AUTO
set accel_dev AUTO
set accel_lpf_type BIQUAD
set accel_lpf_freq 15
set accel_offset_x 0
set accel_offset_y 0
set accel_offset_z 0
set mag_bus AUTO
set mag_dev NONE
set mag_align DEFAULT
set mag_filter_type BIQUAD
set mag_filter_lpf 10
set mag_offset_x 0
set mag_offset_y 0
set mag_offset_z 0
set mag_scale_x 1000
set mag_scale_y 1000
set mag_scale_z 1000
set baro_bus AUTO
set baro_dev NONE
set baro_lpf_type BIQUAD
set baro_lpf_freq 10
set board_align_roll 0
set board_align_pitch 0
[... 203 lines cut ...]
set mix_40 0 0 0
set mix_41 0 0 0
set mix_42 0 0 0
set mix_43 0 0 0
set mix_44 0 0 0
set mix_45 0 0 0
set mix_46 0 0 0
set mix_47 0 0 0
set mix_48 0 0 0
set mix_49 0 0 0
set mix_50 0 0 0
set mix_51 0 0 0
set mix_52 0 0 0
set mix_53 0 0 0
set mix_54 0 0 0
set mix_55 0 0 0
set mix_56 0 0 0
set mix_57 0 0 0
set mix_58 0 0 0
set mix_59 0 0 0
set mix_60 0 0 0
set mix_61 0 0 0
set mix_62 0 0 0
set mix_63 0 0 0
save

rtlopez, 2025-05-14:
problem is here `set pin_i2c_scl 47`, I2C gpio number must be lower than 32.

anas-alali-24, 2025-05-14:
Thanks a lot for your response,  I will change it and try it
May I ask is there any reason to be less than 32? It worked with a custom fFC code 
Thanks again

rtlopez, 2025-05-14:
there is custom I2C driver, bit limitted but optimized.

anas-alali-24, 2025-05-15:
great working perfectly
but I can't get the receiver to work with elrs
I tested with ppm with flysky and I can see everything is fine , but I tried with Elrs (Radiomaster boxer) no luck, any ideas?

[image]

Thanks in advance

rtlopez, 2025-05-15:
show ports tab

rtlopez, 2025-05-15:
settings looks good, check your wiring, remember to cross rx-tx between FC and Receiver, keep in mind that uart3 in configurator is pin_serial_2 in cli.