rtlopez/esp-fc #171: Questions about using ESP32‑S3 Super Mini and NRF24 Mini SMD modules

muki01, 2025-05-18:
First of all, thank you for sharing this awesome project I really appreciate the work you’ve put into it. I have a couple of hardware‑related questions:

1. **ESP32‑S3 Super Mini**  
   * Is this board fully supported?  
   * If so, which pins should I hook up for proper operation?  
   * Can I remap those pins in the firmware or configuration files?

2. **NRF24 Mini SMD module**  
   * I noticed there are NRF24‑related settings does that mean this module is also supported?  
   * Are there any special considerations or limitations I should be aware of?

Thanks in advance for your help!

rtlopez, 2025-05-22:
Hi, I'm glad you enjoy it :)

Most esp32-s3 boards should work. On some you may need to adjust pin assignment. To see what is current setup, go to the cli and type `get pin` command. You can remap all pin functions as you wish. Here is basic CLI reference doc https://github.com/rtlopez/esp-fc/blob/master/docs/cli.md

there is no NRF24 support in this firmware. In place of that you can use esp-now protocol, that can communicate between two esp32 family chips. In the setup guide you can find current limitations https://github.com/rtlopez/esp-fc/blob/master/docs/setup.md

rtlopez, 2025-07-04:
It doesn't work in that way. Esp-now requires handsheke/binding phase to pass, it is for safety. There is a dedicated libarary for this purpose https://github.com/rtlopez/espnow-rclink. Everything happens autmatically, you don't even need to know mac address of any device.

closed 2025-11-11 by rtlopez